hardbyte/python-can #86: provide a way to forward exceptions in the recv-thread

umlaeute, 2016-10-05:
i'm using `socketcan` as backend.

whenever the interface goes down (`ifconfig vcan0 down`), the actual receiver gets an `OSError` exception.
unfortunately this happens in the receiver-thread, and there is no way to be notified of this event.

i'd propose a new (optional) argument `execption_callback` to the `Notifier` class that get's called (with the execption as argument) whenever such an exception is caught.

umlaeute, 2016-10-12:
so after sleeping this over i think that callback is probably an overkill.

instead i would like to propose to just catch "all" exceptions than can be appear while receiving data (e.g. `OSError`) and return `None`, like with other exceptions.

the exceptions should probably be logged...

hardbyte, 2016-10-12:
I suggest we catch the `OSError`s in `Bus.recv` and then throw a `can.CanError`. 

When does `recv` return `None` other than timeout?

Agree that exceptions in the `Notifier` object's  recv thread should be logged.

umlaeute, 2016-10-12:
i'm pretty sure that i had to catch exceptions in two distinct places (using `socketcan_native`) in order to properly deal with a `can0` device that was taken down during operation.

i'll try to be more specific when i have access to the the system again...

closed 2016-10-19 by hardbyte