hardbyte/python-can #107: Split the parsing and the IO

hardbyte, 2016-12-09:
I'm convinced we should clearly split the parsing and the IO for each interface. This obviously effects the entire library - but it should all be internal changes.

See [this http 1.1 example](https://h11.readthedocs.io/en/latest/basic-usage.html) or [this great pycon16 talk](https://www.youtube.com/watch?v=7cC3_jGwl_U).

felixdivo, 2018-02-24:
I just watched that talk you linked yesterday and now get what you meant. ^^ I initially thought about the current `can.io.*` package which didn't make too much sense. 

I totally agree with the need for splitting the parsing and the I/O. Although I think it would mainly benefit the testing & readability of this library, rather than the reusability in other projects. But who knows. 

But there are a few more problems that I personally had with understanding the code of this library, which should be part of this discussion. I think a lot of the project structure could be made quite a bit simpler to make it easier to get started. When I have some time I will open a separate Issue for that to start a discussion on that in general.

felixdivo, 2022-11-16:
@hardbyte Is this also a discussion?