hardbyte/python-can #119: Compatibility with Lawicel CANUSB

remusmp, 2017-01-20:
Is `python-can` compatible with [Lawicel CANUSB](http://www.can232.com/)? What configuration should I use?

hardbyte, 2017-01-20:
It depends if you're running on a reasonably modern linux then yes it is supported via socketcan. 

On windows would take a small amount of work to wrap their DLL. Contributions welcome!

remusmp, 2017-01-20:
Thanks. I'll look into it and wrap their dll.

krixkrix, 2017-05-03:
Did you wrap it or find another python solution for Lawicel CANUSB on windows? I am looking for the same.

remusmp, 2017-05-05:
I started on wrapping it but the thing is Lawicel only provides .net 2.0 dlls for x64. I didn't want my project to depend on .net so my choice was to switch to Linux for the entire project where CAN just worked with no head aches. I found Lawicel CANUSB very outdated from a software point of view.

grodansparadis, 2017-05-08:
The USB2CAN is CANAL and the Lawicel driver should export a CANAL interface (http://www.vscp.org/docs/vscpd/doku.php?id=level1_driver_canusb) but there is also a CANAL interface for the serial channel of it (http://www.vscp.org/docs/vscpd/doku.php?id=level1_driver_can232).

The naming of the "USB2CAN" interface is a bit bad. Should have been "CANAL" IMHO as it is just one device that export this interface (http://www.vscp.org/docs/vscpd/doku.php?id=vscp_daemon_level_i_drivers).

I just found out the existent of python-can today. Great work!

grodansparadis, 2017-05-08:
Sorry for jumping in on the thread btw.

yegorich, 2017-05-11:
Why not just implementing ASCII protocol using python-serial? This way you can handle both serial based devices as also devices talking ASCII protocol via TCP/IP.

PR #195 (2017-10-16): add slcan-support
add slcan/lawicel serial interface
these where supported with socketcan (slcand) already.
But this simple interfaces makes any slcan-compatible device usable for windows also

Note: no need to rename on of current modules any more.

Currently only tested with python2.7.

this should fix #119 and #5

please have a look on it.
I'll do further work on it, if needed.
closed 2017-10-16 by hardbyte