hardbyte/python-can #120: Is Linux socketcan interface not supported?

remusmp, 2017-01-23:
Hi,

I've installed the latest python-can package this way:

`pip3 install git+https://github.com/hardbyte/python-can.git`

In pip list it appears as 2.0.0a1.

Then I typed in these commands in a python terminal:
```
Python 3.5.2 (default, Nov 17 2016, 17:05:23) 
[GCC 5.4.0 20160609] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import can
>>> can.rc['interface'] = 'socketcan'
>>> bus = can.interface.Bus()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.5/dist-packages/can/interfaces/interface.py", line 79, in __new__
    raise NotImplementedError("CAN interface '{}' not supported".format(interface))
NotImplementedError: CAN interface 'socketcan' not supported
```

I'm able to see the CAN bus under Linux via `candump` (I have installed all necessary modules). Here is an example:
```
candump slcan0
  slcan0  001   [4]  1B 00 00 00
  slcan0  002   [2]  38 13
  slcan0  001   [4]  1B 00 00 00
```

hardbyte, 2017-01-23:
Thanks for the report - I'll have a look into it. Would you mind trying with the value `socketcan_native`?

remusmp, 2017-01-24:
This works:
```
import can
can.rc['interface'] = 'socketcan_native'
can.rc['channel'] = 'slcan0'
bus = can.interface.Bus()
```

Once again thanks!

commit aa44f45d (2017-02-01): Choose correct socketcan interface. Closes #120

* Improve handling of configuration.
* If interface is set to socketcan in can.rc['interface'] choose which implementation to use.
* Extract helper function for checking the can.rc global config.
* Slightly rearrange where to find set of valid interfaces.
closed 2017-02-01 by hardbyte