hardbyte/python-can #136: Sent messages are being received on the listener

altran-git, 2017-01-31:
I am using the cyclic example to send a few periodic messages.  I also want to read messages at the same time on the same channel; however, all the messages I am sending are being received on the listener.  Is there a way to disable this?

```
can.rc['interface'] = 'socketcan_native'
can.rc['channel'] = 'can0'
from can.interfaces.interface import Bus
bus = Bus()

msg1 = can.Message(arbitration_id=0x219, extended_id=False, data=[0, 0, 0, 0, 0, 0, 0, 0])
msg2 = can.Message(arbitration_id=0x21A, extended_id=False, data=[0, 0, 0, 0, 0, 0, 0, 0])
msg3 = can.Message(arbitration_id=0x3E, extended_id=False, data=[0, 0, 0, 0, 0, 0, 0, 0])
msg4 = can.Message(arbitration_id=0x512, extended_id=False, data=[0, 0, 0, 0, 0, 0, 0, 0])

task1 = can.send_periodic('can0', msg1, 0.02)
task2 = can.send_periodic('can0', msg2, 0.2)
task3 = can.send_periodic('can0', msg3, 0.2)
task4 = can.send_periodic('can0', msg4, 0.1)

while 1:
        a_listener = can.BufferedReader()
        notifier = can.Notifier(bus, [a_listener])
        m = a_listener.get_message(0.5)
        if m is not None:
            print(m)  
```

hardbyte, 2017-01-31:
Since you're using socketcan you should be able to set this at the device level. Something like this should do it:

```
sudo ip link set can0 type can loopback off
```

I'd recommend reading through the [socketcan docs](https://www.kernel.org/doc/Documentation/networking/can.txt) - perhaps there is a flag we could set when we create the bus?

Also in your example you shouldn't create a new notifier and buffered reader inside your recv loop:

```
a_listener = can.BufferedReader()
notifier = can.Notifier(bus, [a_listener])
while True:
    m = a_listener.get_message(0.5)
    if m is not None:
        print(m)
```

altran-git, 2017-02-01:
Thanks.  I get an Operation not supported when I use loopback off .

What does this mean?  How can I set this driver flag?
In this case the driver flag IFF_ECHO has to be set to prevent the PF_CAN core from locally echoing sent frames (aka loopback) as fallback solution:
    dev->flags = (IFF_NOARP | IFF_ECHO);

prices, 2017-10-05:
The virtual interface also has this problem:

`
import can

can.rc['interface'] = 'virtual'

from can.interfaces.interface import Bus
bus = Bus()

msg1 = can.Message(arbitration_id=0x219, extended_id=False, data=[0, 0, 0, 0, 0, 0, 0, 0])

bus.send(msg1)

m = bus.recv(5)
if m is not None:
    print(m)  

`
Returns:
Timestamp: 1507205077.501877        ID: 0219    000    DLC: 8    00 00 00 00 00 00 00 00

Is there any way to turn this off?  I am having trouble writing unit tests because of this.

christiansandberg, 2017-10-05:
The virtual interface should be fixed in the latest 2.0 pre-release (see #188).

prices, 2017-10-05:
Thanks.  That does indeed fix my problem.

closed 2017-10-19 by hardbyte