hardbyte/python-can #146: modify_data() and then performing a recv() on the same bus will not see the data change.

altran-git, 2017-02-22:
1. Start a task running cyclic messages on a can interface (vcan0)
    **msg = can.Message(arbitration_id=0x123, data=[0, 0, 0, 0, 0, 0], extended_id=False)**
    **task = can.send_periodic('vcan0', msg, 0.20)**

2. Instantiate a bus on the same interface as running task
     **bus = can.interface.Bus(bustype='socketcan_native', channel='vcan0')**

3. Modify data associated with running task on (vcan0)
     **msg.data[0] = 0x99**
     **task.modify_data(msg)**

4.  Perform a recv() on the bus instantiated in step 2.
      **read_msg = bus.recv()**

The read_msg does not see the change performed by the modify task in step 3.  In order for me to see the change, I would have to perform step 2 after step 3.  Any subsequent modify_data calls I will have to reinstantiate a bus and then perform a recv() in order to see the change.

Is this working as intended?

felixdivo, 2018-02-11:
This does not seem to be right. The code responsible for this can be found in `can/interfaces/socketcan/socketcan_native.py`. Any pull requests are welcome. ;) Adding your steps to a unit test in `tests/simplecyclic_test.py` might be helpful when working on this.

christiansandberg, 2018-02-12:
If you keep on reading messages you should get the updated data eventually. The first messages will contain the old data that has already been transmitted.

felixdivo, 2018-02-17:
If this is still not solved, you can reopen this issue or poke us with @mention.

closed 2018-02-17 by felixdivo