hardbyte/python-can #150: Issue with can.send_periodic

EmilianoMaina, 2017-03-18:
Hi,
after few months successfully using python can on my heat pump reading data from CAN Bus in a request / sync receive, I am not able to move to a request / async receive.

I was able to implement the listener. The issue is with periodical sending of data on the bus.

This is the error:

`pi@raspberrypi:~/pyHPSU-master $ python3 pyListener.py
Traceback (most recent call last):
  File "pyListener.py", line 88, in <module>
    task = can.send_periodic('can0', msg, 15.0)
  File "/home/pi/pyHPSU-master/can/broadcastmanager.py", line 75, in send_periodic
    return can.interfaces.interface.CyclicSendTask(channel, message, period)
AttributeError: 'module' object has no attribute 'CyclicSendTask'
`
And this is the corresponding source:

`#!/usr/bin/env python
# -*- coding: utf-8 -*-

import sys, time
import requests
import json
import math
import binascii

import logging
import logging.handlers

logHandler = logging.handlers.TimedRotatingFileHandler("/home/pi/listener.log",when="midnight")
logFormatter = logging.Formatter('%(asctime)s %(message)s')
logHandler.setFormatter( logFormatter )
logger = logging.getLogger()
logger.addHandler( logHandler )
logger.setLevel( logging.INFO )

logging.getLogger("requests").setLevel(logging.WARNING)
logging.getLogger("urllib3").setLevel(logging.WARNING)

try:
    import can
except Exception:
    pass

# Variable Name => String (Costante)
# Signature => Hex (Costante)
# Last Read Timestamp => Time
# Last Publish Timestamp => Time
# Value [0] => Float
# Value [1] => Float
# Value [2] => Float
# Value [3] => Float
# Value [4] => Float
# Last Value => Float (corrisponde a Value [0])
# Average Value => Float (corrisponde alla media aritmetica di Value [x])
# Filtered Value => Float
# Average Filtered Value => Value

dati = [
    [bytes([0x32, 0x10, 0xFA, 0xC0, 0xF6]), "mode",            [0.0, 0.0, 0.0, 0.0, 0.0], 0.0, 0.0],
    [bytes([0x32, 0x10, 0xFA, 0xC0, 0xFB]), "bpv",             [0.0, 0.0, 0.0, 0.0, 0.0], 0.0, 0.0],
    [bytes([0x32, 0x10, 0xFA, 0x06, 0x9B]), "posmix",          [0.0, 0.0, 0.0, 0.0, 0.0], 0.0, 0.0],
    [bytes([0x32, 0x10, 0xFA, 0xC0, 0xF7]), "pump",            [0.0, 0.0, 0.0, 0.0, 0.0], 0.0, 0.0],
    [bytes([0x32, 0x10, 0xFA, 0xC0, 0xFC]), "t_v1",            [0.0, 0.0, 0.0, 0.0, 0.0], 0.0, 0.0],
    [bytes([0x32, 0x10, 0xFA, 0xC0, 0xFD]), "t_dhw1",          [0.0, 0.0, 0.0, 0.0, 0.0], 0.0, 0.0],
    [bytes([0x32, 0x10, 0xFA, 0xC0, 0xFE]), "t_vbh",           [0.0, 0.0, 0.0, 0.0, 0.0], 0.0, 0.0],
    [bytes([0x32, 0x10, 0xFA, 0xC1, 0x00]), "t_r1",            [0.0, 0.0, 0.0, 0.0, 0.0], 0.0, 0.0],
    [bytes([0x32, 0x10, 0xFA, 0xC0, 0xF9]), "ehs",             [0.0, 0.0, 0.0, 0.0, 0.0], 0.0, 0.0],
    [bytes([0x62, 0x10, 0x04, 0x00, 0x00]), "t_hc_set",        [0.0, 0.0, 0.0, 0.0, 0.0], 0.0, 0.0],
    [bytes([0x32, 0x10, 0x03, 0x00, 0x00]), "t_dhw_set",       [0.0, 0.0, 0.0, 0.0, 0.0], 0.0, 0.0],
    [bytes([0x62, 0x10, 0xFA, 0x0A, 0x0C]), "t_ext",           [0.0, 0.0, 0.0, 0.0, 0.0], 0.0, 0.0],
    [bytes([0x32, 0x10, 0xFA, 0xC0, 0xFF]), "t_outdoor_ot1",   [0.0, 0.0, 0.0, 0.0, 0.0], 0.0, 0.0],
    [bytes([0x32, 0x10, 0xFA, 0xC1, 0x05]), "ta2",             [0.0, 0.0, 0.0, 0.0, 0.0], 0.0, 0.0],
    [bytes([0x32, 0x10, 0x13, 0x00, 0x00]), "t_dhw_setpoint1", [0.0, 0.0, 0.0, 0.0, 0.0], 0.0, 0.0],
    [bytes([0x32, 0x10, 0xFA, 0x06, 0x91]), "hyst_hp",         [0.0, 0.0, 0.0, 0.0, 0.0], 0.0, 0.0],
    [bytes([0x32, 0x10, 0xFA, 0x01, 0x12]), "mode_01",         [0.0, 0.0, 0.0, 0.0, 0.0], 0.0, 0.0],
    [bytes([0x32, 0x10, 0xFA, 0x01, 0xDA]), "flow_rate",       [0.0, 0.0, 0.0, 0.0, 0.0], 0.0, 0.0],
]

class CanListener(can.Listener):
    def on_message_received(self, msg):
        for i in range(len(dati)):
            if (msg.data[0:5] == dati[i][0]):
                dati[i][2][4] =  dati[i][2][3]
                dati[i][2][3] =  dati[i][2][2]
                dati[i][2][2] =  dati[i][2][1]
                dati[i][2][1] =  dati[i][2][0]
                dati[i][2][0] =  int.from_bytes(msg.data[5:7], byteorder='big', signed=True)
                dati[i][3]    = msg.timestamp
                logger.info("%50s %05s %05s %f", msg, dati[i][1].ljust(10), dati[i][2], dati[i][3])

try:
    #can_filters = [{"can_id": "0x1c1"}]
    bus = can.interface.Bus(channel='can0', bustype='socketcan_native')
except Exception:
    logger.error ("exception open bus")
    sys.exit(0)

a_listener = CanListener()
notifier = can.Notifier(bus, [a_listener])

logger.info("Can Logger Started")

msg = can.Message(arbitration_id=0x190, data=dati[14][0], extended_id=False, dlc=5)
task = can.send_periodic('can0', msg, 15.0)

try:
    while True:
        time.sleep(15)
        power = 1.163 * float(dati[17][2][0]) * 0.1 * ( float(dati[6][2][0]) - float(dati[7][2][0]) + 0.379 + 0.111)
        print(power)
except KeyboardInterrupt:
    bus.shutdown()
`

I am working with Raspberry + PiCAN2 board. Python version 3.4.2

How could I further troubleshoot ?

EmilianoMaina, 2017-03-18:
Realized that I wasn't using latest python can release. Upgraded but changed only the error:

`Traceback (most recent call last):
  File "pyListener.py", line 88, in <module>
    task = can.send_periodic('can0', msg, 15.0)
  File "/home/pi/pyHPSU-master/can/broadcastmanager.py", line 75, in send_periodic
    return can.interfaces.interface.CyclicSendTask(channel, message, period)
AttributeError: 'module' object has no attribute 'CyclicSendTask'

{'interface': 'socketcan_native', 'channel': 0}
Traceback (most recent call last):
  File "pyListener.py", line 88, in <module>
    task = can.send_periodic('can0', msg, 15.0)
  File "/usr/local/lib/python3.4/dist-packages/can/broadcastmanager.py", line 72, in send_periodic
    return can.interface.CyclicSendTask(channel, message, period)
  File "/usr/local/lib/python3.4/dist-packages/can/interfaces/interface.py", line 99, in __new__
    return cls(channel, *args, **kwargs)
  File "/usr/local/lib/python3.4/dist-packages/can/interfaces/socketcan_native.py", line 150, in __init__
    super(CyclicSendTask, self).__init__(channel, message, period)
  File "/usr/local/lib/python3.4/dist-packages/can/interfaces/socketcan_native.py", line 137, in __init__
    self.bcm_socket = create_bcm_socket(channel)
  File "/usr/local/lib/python3.4/dist-packages/can/interfaces/socketcan_native.py", line 122, in create_bcm_socket
    s = socket.socket(socket.PF_CAN, socket.SOCK_DGRAM, socket.CAN_BCM)
  File "/usr/lib/python3.4/socket.py", line 123, in __init__
    _socket.socket.__init__(self, family, type, proto, fileno)
OSError: [Errno 93] Protocol not supported`

EmilianoMaina, 2017-03-18:
And this is how can interfface is up:

`sudo /sbin/ip link set can0 up type can bitrate 20000`

hardbyte, 2017-03-18:
Hi Emiliano

I'm aware of a few issues with `send_periodic` so I've been working on a new version - hopefully with a cleaner api too. Could you try out the branch [feature-periodic-send-api](https://github.com/hardbyte/python-can/tree/feature-periodic-send-api).

Note the code will have to change to:

```
task = bus.send_periodic(msg, 0.20)
```

Please let me know how you get on. I've tested periodic sending with my PiCAN2 board with Python3.5

EmilianoMaina, 2017-03-18:
Thanks hardbyte.
Just for my clarity: should I download the full branch and substitute current version on my Raspberry ?

EmilianoMaina, 2017-03-18:
I downloaded the branch and copied the can directory.
Changed the code.

But I still have the same error:

`Traceback (most recent call last):
  File "pyListener.py", line 95, in <module>
    task = bus.send_periodic(msg, 15.0)
  File "/home/pi/pyHPSU-master/can/interfaces/socketcan/socketcan_native.py", line 448, in send_periodic
    task = CyclicSendTask(self.channel, msg, period)
  File "/home/pi/pyHPSU-master/can/interfaces/socketcan/socketcan_native.py", line 202, in __init__
    super(CyclicSendTask, self).__init__(channel, message, period, duration=None)
  File "/home/pi/pyHPSU-master/can/interfaces/socketcan/socketcan_native.py", line 181, in __init__
    self.bcm_socket = create_bcm_socket(channel)
  File "/home/pi/pyHPSU-master/can/interfaces/socketcan/socketcan_native.py", line 126, in create_bcm_socket
    s = socket.socket(socket.PF_CAN, socket.SOCK_DGRAM, socket.CAN_BCM)
  File "/usr/lib/python3.4/socket.py", line 123, in __init__
    _socket.socket.__init__(self, family, type, proto, fileno)
OSError: [Errno 93] Protocol not supported
`

christiansandberg, 2017-03-18:
Sounds like an issue with your Linux. Which dist are you using?

EmilianoMaina, 2017-03-18:
`Linux raspberrypi 4.4.38-v7+ #938 SMP Thu Dec 15 15:22:21 GMT 2016 armv7l GNU/Linux`

EmilianoMaina, 2017-03-19:
Upgraded Python to the latest release available but nothing changed:

`Traceback (most recent call last):
  File "pyListener.py", line 95, in <module>
    task = bus.send_periodic(msg, 15.0)
  File "/home/pi/pyHPSU-master/can/interfaces/socketcan/socketcan_native.py", line 448, in send_periodic
    task = CyclicSendTask(self.channel, msg, period)
  File "/home/pi/pyHPSU-master/can/interfaces/socketcan/socketcan_native.py", line 202, in __init__
    super(CyclicSendTask, self).__init__(channel, message, period, duration=None)
  File "/home/pi/pyHPSU-master/can/interfaces/socketcan/socketcan_native.py", line 181, in __init__
    self.bcm_socket = create_bcm_socket(channel)
  File "/home/pi/pyHPSU-master/can/interfaces/socketcan/socketcan_native.py", line 126, in create_bcm_socket
    s = socket.socket(socket.PF_CAN, socket.SOCK_DGRAM, socket.CAN_BCM)
  File "/usr/local/lib/python3.6/socket.py", line 144, in __init__
    _socket.socket.__init__(self, family, type, proto, fileno)
OSError: [Errno 93] Protocol not supported
`

EmilianoMaina, 2017-04-01:
Hello, any hints on my problem ? Thanks !

altran-git, 2017-04-05:
I had the same issues with RaspberryPi.  I've posted my findings at this thread:

https://bitbucket.org/hardbyte/python-can/issues/60/problem-using-cyclic-message

EmilianoMaina, 2017-04-07:
In fact I didn't to anything different from what described above but retrying now after a few days retrying is now working. Thanks for the support !

closed 2017-04-07 by EmilianoMaina