hardbyte/python-can #154: Using python-can with Kvaser on linux

CatherineH, 2017-04-03:
I am trying to get python-can working with a Kvaser leaf on ubuntu 16.04.

I've installed the modules supplied from kvaser.

I'm using this code:

```
from can.interfaces.kvaser.canlib import KvaserBus

interface = KvaserBus(0)
```

It results in this error:

```
Traceback (most recent call last):
  File "pycan-test.py", line 3, in <module>
    interface = KvaserBus(0)
  File "/usr/local/lib/python2.7/dist-packages/can/interfaces/kvaser/canlib.py", line 361, in __init__
    self.set_filters(can_filters)
  File "/usr/local/lib/python2.7/dist-packages/can/interfaces/kvaser/canlib.py", line 413, in set_filters
    canSetAcceptanceFilter(handle, can_id, can_mask, ext)
  File "/usr/local/lib/python2.7/dist-packages/can/interfaces/kvaser/canlib.py", line 39, in _unimplemented_function
    raise NotImplementedError('This function is not implemented in canlib')
NotImplementedError: This function is not implemented in canlib
```

Can someone point out what I'm doing wrong? Should I not be using this interface on linux?

CatherineH, 2017-04-03:
Ah, this is because I didn't install the kvaser linux python library:

```
wget http://www.kvaser.com/software/7330130981911/V1_1_23/canlib-1.1.23.zip
pip install canlib-1.1.23.zip
```

As well as the kvaser linux drivers. I think the links in the documentation need to be updated. I'll make a MR.

hardbyte, 2017-04-03:
Hi Catherine, 
The Kvaser interface on linux was actually the original reason this whole library was written so it should absolutely work.

The problem appears to be more specific - the `set_filters`  code is comparatively new and most likely hasn't been tested on linux with kvaser on python 2. I'm afraid I don't have access to any Kvaser hardware anymore so might need some help to fix this.

For now if you're not using filters you can probably comment out the offending [line](https://github.com/hardbyte/python-can/blob/develop/can/interfaces/kvaser/canlib.py#L414).

christiansandberg, 2017-04-04:
I know that some functions in Kvaser's API are not implemented in Linux, but it's not mentioned in the documentation so I don't know which ones. I can change the code so that filtering is simply ignored if that is not implemented in the driver.

Also, doesn't SocketCAN support Kvaser as well?

CatherineH, 2017-04-04:
Yes! I've been successful using the ctypes implementation of socketcan with python 2. Since that works, I'm not as concerned about this issue.

If I get some free time I'll try to dig into it.

jackm, 2017-06-08:
FYI, this issue is still valid and this bug still exists. I'm not sure why it was decided to close this prematurely just because the user who reported it didn't care about it any more.

I too am attempting to use python-can with python3 on Linux with a [Kvaser Leaf Light v2](https://www.kvaser.com/product/kvaser-leaf-light-hs-v2/) tool. Every time I try to run one of the example scripts, I still run into an exception where it tells me that the "canSetAcceptanceFilter" function isn't implemented in the canlib library from Kvaser.

This is disappointing as I wanted to use python-can to help log CAN message using the Kvaser tool on a Linux platform.

christiansandberg, 2017-06-08:
Are you using the latest develop branch? I closed it because I thought I fixed it.

jackm, 2017-06-08:
Yes, I cloned the develop branch today and then attempted to run the `send_one.py` example script. I made one small change to the script by adding in parameters `bustype='kvaser'` and `channel='0'` to the *can.interface.Bus* constructor call.

I then get an exception when it creates the new Bus object:
`can.interfaces.kvaser.canlib.CANLIBError: Function canSetAcceptanceFilter failed - Not Implemented`

christiansandberg, 2017-06-08:
Okay then I understand what the problem is. The function is implemented in the canlib library but always returns an error. We should also catch CANLIBError when attempting to call this function.

christiansandberg, 2017-06-09:
I don't have the ability to test my solution. @jackm, if it doesn't work please re-open again.

closed 2017-06-09 by christiansandberg

jackm, 2017-06-09:
Confirmed that it now properly catches the exception and the script reports that it was able to send a CAN message on the Kvaser Leaf Light v2

ChrisAlphabet, 2017-10-17:
Hi all,

I am trying to get a project with a Kvaser Leaf Light up and running.
I have installed the drivers and am able to run the examples that came with them.

Other projects I am working on have used the socketcan interface (using a RaspberryPi and a PiCAN2 hat) and I was hoping to use socketcan for the Kvaser aswell.

My question is, how do I bring the interface up so that the Kvaser Leaf shows up in `ifconfig`?

Is there some sort of equivalent to 

```shell
sudo modprobe vcan
# Create a vcan network interface with a specific name
sudo ip link add dev vcan0 type vcan
sudo ip link set vcan0 up
```

that I need to use?

bauer-san, 2017-12-22:
@ChrisAlphabet I think the driver should be loaded when you plug in the Leaf Light.
You can confirm it by:
$ lsmod

Mine shows Module leaf and kvcommon.

Geordie-Jon, 2023-08-08:
Hey - this is closed but 
kvaser publish their api - [Kvaser](https://www.kvaser.com/canlib-webhelp/)

on linux you may also use the function canAccept(handle, envelope, flag)
its just a 2 step of setAcceptanceFilter()
first set code
then set mask

works on both windows and linux

commit ee844e64 (2017-04-09): Kvaser should not fail if filtering is not supported

Linux API currently does not support canSetAcceptanceFilter
Fixes #154
commit 091e6b0c (2017-06-09): Catch errors from kvaser driver when setting filters

Fixes #154.