hardbyte/python-can #156: Broadcast Manager Error

mbmuldoon, 2017-04-03:
In attempting to use the can.send_periodic with socketcan_native using the develop branch running python 3.4.2 I get the following result:
>>> task = can.send_periodic('vcan0', msg, 0.20)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/lib/python3.4/site-packages/can/broadcastmanager.py", line 72, in send_periodic
    return can.interface.CyclicSendTask(channel, message, period)
  File "/usr/lib/python3.4/site-packages/can/interface.py", line 101, in __new__
    return cls(config['channel'], *args, **kwargs)
  File "/usr/lib/python3.4/site-packages/can/interfaces/socketcan/socketcan_native.py", line 164, in __init__
    self._tx_setup(message)
  File "/usr/lib/python3.4/site-packages/can/interfaces/socketcan/socketcan_native.py", line 173, in _tx_setup
    self.bcm_socket.send(header + frame)
OSError: [Errno 22] Invalid argument

hardbyte, 2017-04-03:
This has (slowly) been worked on - would you mind trying with the [feature-periodic-send-api](https://github.com/hardbyte/python-can/tree/feature-periodic-send-api) branch?

mbmuldoon, 2017-04-04:
Thanks the feature-periodic-send-api branch works.

closed 2017-04-04 by mbmuldoon