hardbyte/python-can #157: Software filtering of messages

christiansandberg, 2017-04-04:
Some interfaces support multiple filters, some support only one filter, and some do not support filtering at all or uses ranges instead of ID and mask (PCAN).

It could reduce the burden of the application if python-can filtered out messages according to the configured filtering in software. I [did this in the kvaser interface](https://github.com/hardbyte/python-can/blob/1.5.2/can/interfaces/kvaser/canlib.py#L448) before but I think it's better to implement something that all interfaces can use.

Suggestion is to implement this in either the BusABC class or in the utils package so that specific interface implementations can take advantage of them.

felixdivo, 2018-02-11:
It could be implemented in `BusABC` and some interfaces could override this if they are able to do it in the OS kernel or in hardware.

felixdivo, 2018-02-23:
I have thought a little bit about this and want to propose this new `BusABC` implementation:

```Python

from abc import ABCMeta, abstractmethod

class BusABC():
    """A sketch to show the suggested design.

    The comments/docs/implementation are of course shortened
    to make it more readable. In the release this would be
    documented way better.

    New interfaces would need to implement:

        * __init__()
        * _recv_internal()
        * send()
        * set channel_info atribute

    That is pretty much the same as perviously, only that
    recv() is now called _recv_internal() and must return
    `(msg, False)` instead of only `msg` if it didn't support
    filters before and `(msg, True)` if it did.

    It doesn't change *ANY* method signatures used by the actual
    users of the library, and only requires minimal changes in
    the infterfaces.
    """

    @abstractmethod
    def __init__(self, channel=None, can_filters=None, **config):
        """Same as always, except it now has to call self._apply_filters()
        insteat of setting the field itself."""
        self._apply_filters(can_filters)

    def recv(self, timeout=None):
        """Not abstract any more."""

        msg, already_filtered = self._recv_internal(timeout=timeout)

        if msg and (already_filtered or self._matches_filter(msg)):
            return msg

        else:
            # discard message & try next one if there still is time

            elapsed_time = 0 # actually measure this in the real implementation

            if elapsed_time >= timeout:
                return None
            else:
                # is a call stack overflow a real scenario here?
                return self.rcev(timeout=timeout - elapsed_time)

    @abstractmethod
    def _recv_internal(self, timeout=None):
        """The old implementation of recv() moves to here.
        This method returns a tuple (msg, already_filtered)
        containing the message that was rceived (or None) and
        whether it was already filtered by the bus (for example
        in a kernel driver or even in real hardware).
        """
        pass

    def _matches_filter(self, msg):
        """Checks whether the given message matches one of the current filters."""
        for filter in self.filters:
            return True # do the actual checking
        # else
        return False

    def set_filters(self, filters):
        """Not pseudo-abstract any more.
        Now ALL interfaces support filters."""
        self.filters = filters
        self._apply_filters(filters)

    # These methods/fields remain exactly as they are
    def __str__(self): pass
    def send(self, msg): pass # still @abstractmethod
    def send_periodic(self, msg): pass
    def __iter__(self): pass
    def flush_tx_buffer(self): pass
    def shutdown(self): pass
    channel_info = 'unknown'
    __metaclass__ = ABCMeta

```

About the rather unusual return value of `_recv_internal() -> (msg, already_filtered)`. There were two approaches that came to my mind: 

1.) The `BusABC` class could have a private attribute (maybe `_supports_low_level_filtering`) that is `True` if the bus implementation supports kernel level or hardware level filters and else is set to `False`. And based on that `recv()` could figure out whether it has to filter in software or whether filtering has already been done.

2.) The approach I have shown above is a bit more flexible since it might for example allow normal IDs to be filtered in hardware but not extended IDs. 

What do you think? @christiansandberg @hardbyte

felixdivo, 2018-02-24:
Closes #104.

closed 2018-05-28 by felixdivo

Bobet21, 2018-05-28:
Can you guys provide a sample code of how you handle filtering on can2usb device? By the way thanks a lot for your work :+1:

felixdivo, 2018-05-28:
Here is the current implementation (=the implementation for the next version): https://github.com/hardbyte/python-can/tree/develop/can/interfaces/usb2can

PR #277 (2018-05-28): Software filtering for all Buses
This basically implements #157 (with some small changes). 

The `BusABC` class is now finished, and I want to ask whether that is looking good. If it is, I will adjust the other interfaces one by one to make it work with this new frontend. It shouldn't be too much work though. It would be nice if someone could comment on the docstrings / the semantics of the methods.

Well, and the tests will obviously fail for now.

I also did some refactoring in interface.py so it will be quite easy to add support for #51 later.

Closes #157. Closes #104.