hardbyte/python-can #159: Sleep precision in broadcastmanager and player

cowo78, 2017-04-12:
Currently the broadcastmanager uses time.time and time.sleep as scheduler functions. To get better timing precision (depending also on the kernel and python versions) it may be worth using clock_nanosleep and a monotonic clock (time.clock_gettime(CLOCK_MONOTONIC)).
Of course we're not talking about hard real time but it should be a nice addon without much trouble. I can provide some sample code or a tentative implementation but currently I don't have any CAN hardware to test it.

christiansandberg, 2017-04-16:
I assume you are referring to the thread based fallback in the new cyclic API branch. Would that be possible through ctypes or is compiling a C-module needed? On the other hand I think most Linux users use SocketCAN in which case this solution is not used anyway.

cowo78, 2017-04-18:
Correct; feature-periodic-send-api branch, broadcastmanager.ThreadBasedCyclicSendManager and player.MessageSync classes.
That would be ctypes stuff.
You're probably right, kernel BCM is already using hrtimer, so the plus would be limited. Feel free to close/reject/whatever.

closed 2017-08-24 by christiansandberg