hardbyte/python-can #162: yocto image: canplayer in setup.py clashes with

royemmerich, 2017-05-10:
The Linux `can-utils` package already has a [`canplayer`](https://github.com/linux-can/can-utils) binary which clashes with the `canplayer` entry point in the latest [`setup.py`](https://github.com/hardbyte/python-can/blob/develop/setup.py#L31).

Trying to compile my yocto image results in the following error:

```
Collected errors:
 * check_data_file_clashes: Package can-utils wants to install file /home/phyvm/yocto_builds_PD16.2.0/build/tmp/work/phyboard_regor_am335x_1-phytec-linux-gnueabi/phytec-headless-image/1.0-r0/rootfs/usr/bin/canplayer
	But that file is already provided by package  * python-can
 * opkg_install: Cannot install package packagegroup-hwtools.
```

christiansandberg, 2017-05-13:
What should it be named instead? Maybe canlogplayer?

royemmerich, 2017-05-13:
what about `pycanplayer` or `python_canplayer`?

hardbyte, 2017-05-14:
What about removing the scripts and just having them as modules?

```
$ python -m can.player
$ python -m can.logger
$ python -m can.server
```

christiansandberg, 2017-05-14:
I personally prefer that way since I have control over which interpreter is used. But it is harder to remember.

PR #165 (2017-06-16): Stop scripts conflicted with can-utils
Removes the installed scripts (`canlogger` etc) and exposes them as modules:

```
$ python -m can.player
$ python -m can.logger
$ python -m can.server
```

Fixes #162
commit 0bba2127 (2017-06-16): Scripts conflicted with can-utils

Fixes #162
closed 2017-06-16 by hardbyte