hardbyte/python-can #173: Support CAN-FD

stephen-park, 2017-07-25:
The current car CAN specification is moving to FD. 
FD is also supported in Kvaser's library. It would be very useful if you could add FD's functionality to a very powerful Python-can library

christiansandberg, 2017-07-26:
I'm not an expert in CAN-FD, but I agree it should definitely be implemented here. Here is a proposal of the interface:

* Add `fd` and `data_bitrate` arguments when constructing the bus for specifying that FD should be supported and which bitrate to use for the data transfer.
* Add `is_fd`, `bitrate_switch` and `error_state_indicator` arguments and attributes to the Message class.
* If `is_fd` is `True`, data length should be one of 12, 16, 20, 24, 32, 48 or 64.

Nanahara, 2017-09-08:
> If is_fd is True, data length should be one of 12, 16, 20, 24, 32, 48 or 64.

In FD, the data length beetween 0 and 8 bytes is valid too (0000b - 1000b). Next, the length (12, 16, 20, 24, 32, 48 and 64) is coded like this :
- 1001b -> 12
- 1010b -> 16
- 1011b -> 20
- etc

christiansandberg, 2017-09-08:
Thanks for pointing that out!

I will probably not be able to do anything on this in the near future unfortunately.

skpang, 2017-12-18:
I'm happy to help out with the testing of CAN-FD. I've got a Microchip MCP2517FD board for the Raspberry Pi and a kernel build with SocketCAN supporting CAN-FD. I also have a PCAN-USB Pro FD from Peak System used for testing.

christiansandberg, 2017-12-19:
Cool, there is a branch called can-fd if you want to test SocketCAN. I haven’t implemented it for Peak backend.

PR #207 (2018-01-15): Add CAN-FD support
Initial support for socketcan_native and kvaser. Tested on virtual CAN only.
Experimental support for BLF logger
Only warn on DLC too long since some loggers support long data lengths but not FD frames

Fixes #173.
closed 2018-01-15 by christiansandberg