hardbyte/python-can #176: Python 3.4, socketcan_native:  Network is down (it's not)

MrDadaGuy, 2017-08-03:
shell script:

`sudo ip link set can0 down
sudo ip link set can0 type can restart
sudo ip link set can0 type can bitrate 500000 
sudo ip link set can0 up
`
Python code:
`    class CanListener(can.Listener):
        def on_message_received(self, msg):
            try:
                queue_out.put("h\t" + str(msg))
            except:
                exc_type, exc_value, exc_traceback = sys.exc_info()
                logger.error(''.join('!! ' + line for line in traceback.format_exception(exc_type, exc_value, exc_traceback)))
     

    class TextWriter(can.ASCWriter):
        def __init__(self):
            super().__init__("CANASCII.ASC")       
        pass

    try:
        logger.debug('Configuring can0...')
        subprocess.Popen(["/opt/car-telemetry/env/canbus/canconfig500k.sh"], shell=True)
    
        logger.debug ('Starting CAN HIGH...')
    
        can_filters = []  #  [{"can_id": "0x1c1"}]
        bus = can.interface.Bus(channel="can0", bustype="socketcan_native", can_filters=can_filters)    # was socketcan_ctypes  socketcan_native
        a_listener = CanListener()
        b_listener = TextWriter()
        notifier = can.Notifier(bus, [a_listener, b_listener])
    except:
        exc_type, exc_value, exc_traceback = sys.exc_info()
        logger.error(''.join('!! ' + line for line in traceback.format_exception(exc_type, exc_value, exc_traceback)))
    `

Result:
```
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python3.4/threading.py", line 920, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.4/threading.py", line 868, in run
    self._target(*self._args, **self._kwargs)
  File "/usr/local/lib/python3.4/dist-packages/can/notifier.py", line 39, in rx_thread
    msg = self.bus.recv(self.timeout)
  File "/usr/local/lib/python3.4/dist-packages/can/interfaces/socketcan_native.py", line 341, in recv
    packet = capturePacket(self.socket)
  File "/usr/local/lib/python3.4/dist-packages/can/interfaces/socketcan_native.py", line 274, in capturePacket
    cf, addr = sock.recvfrom(can_frame_size)
OSError: [Errno 100] Network is down
```

felixdivo, 2017-12-12:
@MrDadaGuy Have you made any progress since august? How did you solve it?

felixdivo, 2017-12-12:
OK nice, thank you :)

closed 2018-05-22 by felixdivo