hardbyte/python-can #186: VectorBus: permissionMask should be set before calling xlOpenPort

oomzay, 2017-09-12:
@ Version 2.0.0b1

The permissionMask parameter is both input & output to xlOpenPort and must be set before the call to xlOpenPort in order to request "init access" permission.  This is necessary, for example, if the bitrate is to be changed later with a call to xlCanSetChannelBitrate.

I.e. In class VectorBus...

  ```
   permission_mask = vxlapi.XLaccess()
   # HERE: Set mask to request init permission if needed
   if bitrate:
       permission_mask.value = self.mask   
   vxlapi.xlOpenPort(self.port_handle, self._app_name, self.mask,
                          permission_mask, rx_queue_size, ...
```

christiansandberg, 2017-09-12:
Makes sense. I'll try to fix it soon unless you could create a pull request?

oomzay, 2017-09-12:
'tis done.

closed 2017-09-12 by christiansandberg