hardbyte/python-can #189: support for ETAS INCA

EVYang1992, 2017-09-15:
is there any ideas for ETAS INCA integrating?

EVYang1992, 2017-09-15:
i have a hardware INCA 592, and wanna make a flashing software, but it seems like till now the python-can doesn't support ETAS API, i think ETAS is commonly used  nowadays, anybody any ideas for this?

zorce, 2017-12-08:
Is this really relevant to this project? ETAS has it's own COM interface you can remote control it with pywin32 COM interface [here](http://docs.activestate.com/activepython/3.3/pywin32/html/com/win32com/HTML/QuickStartClientCom.html).

felixdivo, 2018-04-30:
Well, `python-can` could build on `pywin32` and abstract a little bit from the underlying layers.

hardbyte, 2018-04-30:
Yeah I can't see any reason why not - still would make sense for the application layer to use the same Message types etc. 

Creating a new interface isn't too hard - you can even do it as a plugin now. If you (or anyone else) is interested take a look at some of the implemented interfaces in https://python-can.readthedocs.io/en/2.1.0/interfaces.html

fanfanruyun, 2019-01-03:
I use pythonnet ,the com-api of inca,that can control ,if we have a dbc file,we can use internal of inca.