hardbyte/python-can #191: IXXAT Backend: "CAN acknowledgment error" repeated infinitely if slave not available.

jrast, 2017-10-03:
If I try to send a message to a slave which is not yet ready to process CAN messages, the following message is printed over and over again: 

    WARNING:can.ixxat:CAN acknowledgment error

Backend: IXXAT
Python-CAN Version : 2.0.0-beta.1
Windows: Windows 7 Professional, 64bit

cowo78, 2017-10-06:
As a quick fix you can use the non-blocking version (timeout=0). IMHO the ACK error should be signalled somehow, agree that the zillion log messages while the device boots is not useful.

felixdivo, 2018-02-17:
But isn't that exactly what ACK frames are for? Detecting when some "slave" is not getting the message.

felixdivo, 2018-06-14:
This behaviour seems to be intended. Please reopen / send a PR if this still annoys you.

closed 2018-06-14 by felixdivo