hardbyte/python-can #201: Possible Bug in IXXAT Implementation

jrast, 2017-11-10:
I'm not quite shure if it's an actual bug or a missunderstanding on my side.

If we look at the main part of the [IXXATBus.recv() method](https://github.com/hardbyte/python-can/blob/develop/can/interfaces/ixxat/canlib.py#L387), mainly at the part which handles the timeout != 0 case ([Line 408 ff](https://github.com/hardbyte/python-can/blob/develop/can/interfaces/ixxat/canlib.py#L408)):

```python
# Wait if no message available
t0 = _timer_function()
elapsed_ms = 0
remaining_ms = 0
while elapsed_ms <= timeout:
	elapsed_ms = int((_timer_function() - t0) * 1000)
	remaining_ms = timeout - elapsed_ms
	# Wait until at least one frame is in the buffer
	try:
		_canlib.canChannelWaitRxEvent(self._channel_handle, remaining_ms)
	except VCITimeout:
		log.debug('canChannelWaitRxEvent timed out after %dms', remaining_ms)
		return None

	# In theory we should be fine with a 0 timeout since the rxEvent was already
	# set but I've seen timeouts appearing here and there
	try:
		_canlib.canChannelReadMessage(self._channel_handle, 0, ctypes.byref(self._message))
	except VCITimeout:
		continue

	# See if we got a data or info/error messages
	if self._message.uMsgInfo.Bits.type == constants.CAN_MSGTYPE_DATA:
		tm = _timer_function()
		break

	elif self._message.uMsgInfo.Bits.type == constants.CAN_MSGTYPE_INFO:
		log.info(CAN_INFO_MESSAGES.get(self._message.abData[0], "Unknown CAN info message code {}".format(self._message.abData[0])))

	elif self._message.uMsgInfo.Bits.type == constants.CAN_MSGTYPE_ERROR:
		log.warning(CAN_ERROR_MESSAGES.get(self._message.abData[0], "Unknown CAN error message code {}".format(self._message.abData[0])))

	elif self._message.uMsgInfo.Bits.type == constants.CAN_MSGTYPE_TIMEOVR:
		pass
	else:
		log.warn("Unexpected message info type")
```

Now consider multiple messages arrive very fast and a RxEvent occours. We start processing the first message, but it happens to be a CAN info or error message. This is fine, we just ignore the data and return to the top of the loop, where we wait for the next RxEvent. In my understanding `canChannelWaitRxEvent` will timeout despite having data in the Queue, therefore `None` is returned instead of a new message.

The same would happen if multiple messages arrive very fast and one readout cycle lasts longer than the time between two messages.

I might be wrong, so please correct me if the behaviour is not as described. If someone can confirm this behaviour, I can provide a fix for this.

christiansandberg, 2017-11-10:
From [the manual](https://www.ixxat.com/docs/librariesprovider8/default-document-library/downloads/vci-v3/vci-v3-api-manual.pdf):

> To check whether the receive event has already occurred without blocking the calling program, the value 0 can be specified in the parameter dwMsTimeout when calling the function.

It sounds like also previous RxEvents are checked when calling canChannelWaitRxEvent. Otherwise using a timeout of 0 would always result in timeout.

christiansandberg, 2017-11-10:
But I don't know why we can't just use canChannelReadMessage's timeout handling instead...

jrast, 2017-11-10:
So you would suggest to remove the canChannelWaitRxEvent code and replace 

```python
# In theory we should be fine with a 0 timeout since the rxEvent was already
# set but I've seen timeouts appearing here and there
try:
	_canlib.canChannelReadMessage(self._channel_handle, 0, ctypes.byref(self._message))
except VCITimeout:
	continue
```
 with something like

```python
try:
	_canlib.canChannelReadMessage(self._channel_handle, remaining_ms, ctypes.byref(self._message))
except VCITimeout:
	return None
```

jrast, 2017-11-10:
I can try this, but I think there is something else wrong in my code which wraps python-can.

PR #203 (2017-12-09): ixxat: In recv(), avoid the use of canChannelWaitRxEvent()
The function canChannelWaitRxEvent() does not always return even if there is one message in the FIFO. This looks like a regression with at least VCI V4 4.0.482 since with VCI v3 3.5.4 there is no problem.

This commit also refactor the recv() function in order to be more robust.
    
While testing this fix, I discover that canChannelReadMessage(), with VCI V4, can randomly return the error code VCI_E_RXQUEUE_EMPTY, which by the documentation should never happen, so catch it...

Fix #201
commit f1bdc053 (2017-12-09): ixxat: In recv(), avoid the use of canChannelWaitRxEvent()

The function canChannelWaitRxEvent() does not always return even if there is one message in the FIFO.
This looks like a regression with at least VCI V4 4.0.482 since with VCI v3 3.5.4 there is no
problem.

This commit also refactor the recv() function in order to be more robust.

While testing this fix, I discover that canChannelReadMessage(), with VCI V4, can randomly return
the error code VCI_E_RXQUEUE_EMPTY, which by the documentation should never happen, so catch it...

Fix #201
closed 2017-12-09 by hardbyte