hardbyte/python-can #202: send and recv at the same time

EVYang1992, 2017-11-13:
when i use one interface(for example: kvaser), but i must use send and recv at the same time, how can i do this, must i use 2 threading or do i have an another easy way? thanks

christiansandberg, 2017-11-22:
Yes you need to receive messages in a separate thread. You can use the [Listener](http://python-can.readthedocs.io/en/latest/listeners.html) and [Notifier](http://python-can.readthedocs.io/en/latest/api.html#notifier) classes to accomplish that.

felixdivo, 2017-12-11:
Isn't this all sorted out? 
See also [this example](https://github.com/hardbyte/python-can/blob/develop/examples/vcan_filtered.py).

EVYang1992, 2017-12-11:
Hi, it contains no support controller for ETAS INCA

Best Regards!/ Mit besten Grüßen!
Yue Yang

christiansandberg, 2017-12-28:
I will close this now unless you have suggestion of some better documentation. INCA support is handled in a different issue.

closed 2017-12-28 by christiansandberg