hardbyte/python-can #205: Buffered sending in socketcan-native

felixdivo, 2017-11-22:
When I send a lot of messages over a hardware CAN bus running at 250.000 bit/s, some problems arise when the messages are sent in a very fast pace:

    for n in range(0, 10000):
        message = can.Message(arbitration_id=42, data=bytearray([1, 2, 3]))
        sender.send(message)

Obviously the CAN bus itself cannot handle all these messages that fast, so its understandable that about 80% of the messages are dropped (this obviously varies a lot). But wouldn't it be great if the library would buffer the send requests in a queue and dispatch them at a rate the can can handle? Because even when sending just 30 messages (a more realistic scenario), somewhere between 2 and 3 messages are dropped. 

There is a dirty solution that I implemented for now:

    for n in range(0, 10000):
        message = can.Message(arbitration_id=42, data=bytearray([1, 2, 3]))
        sender.send(message)
        time.sleep(1.0 / 2000) # sleep a little bit, else the CAN is overwhelmed

But I do not think that this is a task for the user of the library, but should instead be handled by python-can itself. I do not think putting a delay of 0.5 ms in there would be a good idea, but rather a queue on a separate thread, or at lease give the option to do so. And that thread would try to send the messages and retry on failure, dispatching all waiting messages one by one. 

What do you think about this, or is that even implemented somewhere?

christiansandberg, 2017-11-22:
One thing we can do is to support the timeout argument in the send method. By using [select.select](https://docs.python.org/3/library/select.html#select.select) we can make sure that it is possible to queue another message and if not, the method will block until it is or timeout occur. This is not possible on all interfaces however and the behaviour can be a bit different.

felixdivo, 2017-11-22:
Oh, that is a nice way to do it. This still blocks the caller but I guess that is acceptable in most scenarios. One still has to define a timeout for this to work though. That should probably get some attention in the docs, right?

felixdivo, 2017-11-22:
As a side note; when will Milestone 2.0 be pushed to [PyPI](https://pypi.python.org/pypi/python-can), so we can use that patch with `pip`?

christiansandberg, 2017-11-25:
I think if the application sends messages faster than the bus can handle, something must slow it down. Otherwise the application will just keep on going and the buffer will eventually grow very big.

I've described the timeout very shortly in [the docs](http://python-can.readthedocs.io/en/latest/bus.html#can.BusABC.send) but I will change it a little since the behaviour is slightly different for different interfaces.

felixdivo, 2017-12-09:
Is that done by now, so that we can close the issue?

hardbyte, 2017-12-09:
Ok closing. Please reopen if this isn't all sorted out.

closed 2017-12-09 by hardbyte