hardbyte/python-can #208: vector interface raise exception: XL_ERR_WRONG_PARAMETER

daily2008, 2017-12-04:
Hi,
when I use the vector inferface, it raise execption as: 
  File "D:\SVN\linux_svn\trunk\brilliance_m8x\robot\src\lib\TACommonLib\Python_CAN_Library\can\interfaces\vector\canlib.py", line 85, in __init__
    vxlapi.XL_INTERFACE_VERSION, vxlapi.XL_BUS_TYPE_CAN)
  File "D:\SVN\linux_svn\trunk\brilliance_m8x\robot\src\lib\TACommonLib\Python_CAN_Library\can\interfaces\vector\vxlapi.py", line 74, in check_status
    raise VectorError(result, xlGetErrorString(result).decode())
VectorError: XL_ERR_WRONG_PARAMETER

I have already use PCAN interface and it works ok. So I just change following settings for my scripts:
1) change parameter interface="vector", channel="0,1,2,3"
2) connect Vector CAN/LIN interface VN1640A hardware to PC.
So are there any else problems for my settings? I have read the "XL Driver Library - Description.pdf", it mentioned need do some configuration with "Vector hardware config tool", I'm not sure these configurations must be done manually before running scripts?
thank you very much!

christiansandberg, 2017-12-04:
You should run Vector hardware config tool and check the configuration for CANalyzer, if it exists. Otherwise you must create an application, assign your channels, then specify `app_name="Your application"` to your Bus constructor.

daily2008, 2017-12-05:
it works ok now. thanks!

closed 2017-12-05 by daily2008