hardbyte/python-can #209: BLFReader is really slow

zorce, 2017-12-08:
Hi,

I'm currently using the BLFReader in a project to read alot of log files to do some anasyzis on them bundled with can4python for KCD database support. But it takes so mush time to run thru the blf files. Is it possible to multiprocess this someway to speed it up? Cannot just read one LOBJ per process because the blocks could have overlapping data.

Br Dennis

christiansandberg, 2017-12-08:
I did some minor changes a few days ago to get rid of some data copies, but it will probably not make a huge difference. Multiprocessing poses a number of challenges as you already mentioned.

christiansandberg, 2017-12-09:
Besides, if you have multiple log files, can’t you just process them concurrently?

zorce, 2017-12-11:
Yes, running multiple log files is one way to do it but currently i'm faced with around 4tb of log files :lollipop:

christiansandberg, 2017-12-28:
I’m closing this for now unless there is some simpler way to speed it up. It’s probably not many people that have 4 TB of log files to analyze all at once. Some suggestions to try:

* Process multiple log files in separate processes
* Try a more optimized Python implementation like PyPy
* Use a compiled language like C++ and Toby Lorenz’ library
* Rule out other bottlenecks like debug prints, network, processing of messages etc

closed 2017-12-28 by christiansandberg