hardbyte/python-can #215: Can write a multiprocessingBaseCyclicSendManager?

daily2008, 2017-12-15:
Hi,
Now there is a ThreadBasedCyclicSendManager, it start scheduler.run in a thread. From my testing, if there are some other python parallel thread running(e.g. start a listener or anything else python scripts), the period time of sending messages is not accurate (from my testing, one message with cycle 25ms, sometimes it may take more than 150ms delay, that is not acceptable!) Maybe use multiprocessing will imporve the performance. thanks!

hardbyte, 2017-12-15:
Ok thanks for the observation. If you do experiment with the `multiprocessing` module please let us know the results.

The ideal situation is to delegate any cyclic tasks to the CAN hardware, or to the operating system kernel. Does your system support periodic tasks in hardware?

daily2008, 2017-12-15:
I used win10.  it seems use multiprocessing is difficult, the main problem is how to share the memory(created task and bus) between two process. @hardbyte, do you have any good idea?

TravisJoe, 2017-12-15:
I have used multiprocessing with CAN before. But mostly with CAN as it's own processes and then passes information back to a data manager processes.

By far Queues are the easies method.

daily2008, 2017-12-15:
multiprocessing.manager() can only transfer simple data, such as dict, list. It seems cannot transfer instance if need make changes in another process with this instance. I write a test scripts as below, it cannot works. 

import multiprocessing
import threading
import sched
import time

class task:
    def __init__(self):
        self.message = {'CAN_ID': '317', 'ASC_CAN_DATA': '\x01\x02\x00\x00\x02\xFF\xFF\x00'}
        self.next_time = time.clock()

def func(scheduler, dct):
    task = dct['317']
    task.next_time += 0.5
    scheduler.enterabs(task.next_time, 0, func, (scheduler, dct))
    
if __name__ == '__main__':
    scheduler = sched.scheduler(time.clock, time.sleep)
    temp_time = time.clock()
    manager = multiprocessing.Manager()
    dct = manager.dict()
    task = task()
    dct['317'] = task
    #p = threading.Thread(target=scheduler.run)
    p = multiprocessing.Process(target=scheduler.run)
    p.daemon = True
    scheduler.enterabs(task.next_time, 0, func, (scheduler, dct)) 
    p.start()    
    time.sleep(2)

TravisJoe, 2017-12-15:
I could be wrong, but because processes essentially are completely isolated processes  they cannot share instances of things like CAN. I feel like I tried that before and realized it needs to live all in the same process. That is when I went to a CAN manager process which only focus was on managing messages. Because it stayed alive I used it more like a thread with a Queue to pass data back and forth. 

For me, passing dicts, tuples and lists was more than enough.

christiansandberg, 2017-12-16:
I think multiple processes will be problematic as some interfaces can not use bus handles created in a separate process.

It could be better to run each message in a separate thread. But it will require a mutex to avoid accessing the bus simultaneously.

christiansandberg, 2017-12-16:
Could you try the latest develop branch and see if it is any better?

As hardbyte mentioned it won't work perfectly on an OS such as Windows which is not suited well for real time performance. Many CAN interface makers have solved this by delegating such tasks to the CAN hardware. Currently IXXAT backend has support for this, but it should be possible to support the more expensive Kvaser and NI-CAN interfaces too.

felixdivo, 2018-02-20:
@daily2008 Have you made any progress on it by now?

closed 2018-03-02 by felixdivo