hardbyte/python-can #219: usb2can with Python3 dead (fix included)

iurly, 2018-01-08:
Hi,
I found a couple of show-stoppers with usb2can using Python3.
Here's my patch:
```
        # set flags on the connection
        if 'flags' in kwargs:
            enable_flags = kwargs["flags"]

        else:
            enable_flags = 0x00000008

        # code to get the serial number of the device
        if 'serial' in kwargs:
            deviceID = kwargs["serial"]
        elif channel is not None:
            deviceID = channel
        else:
            from can.interfaces.usb2can.serial_selector import serial
            deviceID = serial()

        # set baudrate in kb/s from bitrate
        # (eg:500000 bitrate must be 500)
        if 'bitrate' in kwargs:
            br = kwargs["bitrate"]

            # max rate is 1000 kbps
            baudrate = min(1000, int(br/1000))   ########## <------- This was max(...)
        # set default value
        else:
            baudrate = 500

        connector = format_connection_string(deviceID, baudrate)

        self.handle = self.can.open(connector.encode(), enable_flags)   ########### <------- here .encode() was missing
```
Apparently in Python 3 ctypes strings must be passed as bytearray, I don't know if this breaks compatibility with Python 2 though. The above works for me.
Oh, and the max() was clearly wrong.

felixdivo, 2018-02-11:
I think that with Python 2 `str.encode('utf-8')` returns a `str` object, which is equivalent to `bytes` on that version. And with Python 3, `str.encode('utf-8')` returns a `bytes` object which should be fine for the ctypes interface. So that should work. 

I added the fixes in PR #248. @iurly It would be nice if you can confirm that this is what you intended and maybe test it with Python 2 & 3.

iurly, 2018-02-12:
Hi @felixdivo 
yes, commit 0c588b3 is exactly what I meant.
I'm not able to test it though.

commit 7af1021f (2018-02-17): Fix issues in USB2CAN with Python3

Closes #219
closed 2018-02-17 by hardbyte