hardbyte/python-can #233: Changing msg arbitration_id

vargabalint92, 2018-01-25:
Hi everybody,

i have the following problem: 
my little script sending periodically a message and with the same script i would like to read the message. But somehow it does not work, i get a wrong arbitration_id in the on_message_received function (sent id  is 218038402, gotten id is 130).

Do i use the callback in a wrong way? is it a problem of the parent class (can.Listener)?

Thank you for your help!

Balint

`import can

class CanListener(can.Listener):

    def on_message_received(self, msg_get):
        global new_msg
        print('Callback ID ', msg_get.arbitration_id)
        print('Callback Data ', msg_get.data)
        new_msg = True

def main():
    can_interface = 'can0'
    bus = can.interface.Bus(can_interface, bustype='socketcan')

    global msg_send, new_msg

    try:
        msg_send = can.Message(arbitration_id=0xcff0082,
                            is_remote_frame=False,
                            extended_id=True,
                            is_error_frame=False,
                            dlc=8,
                            data=[0xa7, 0x3, 0x21, 0x7e, 0x89, 0x0, 0x0, 0x0])

        task1 = can.send_periodic('can0', msg_send, 5)

        a_listener = CanListener()
        notifier = can.Notifier(bus, [a_listener])
        new_msg = False

        while True:

            if new_msg:
                print('sent msg id ',msg_send.arbitration_id)
                new_msg = False

    except KeyboardInterrupt:
        print("Stop")
        bus.shutdown()
        notifier.stop()

if __name__=='__main__':
    main()
`

felixdivo, 2018-02-23:
Sorry, I simply cannot reproduce your problem. 

Maybe it's inconsistent because under Python 2.7 the `socketcan_ctypes` interface is used and with Python 3.5 `socketcan_native` is used. I pretty sure that `can.Listener` is not the problem, because [it's source code](https://github.com/hardbyte/python-can/blob/develop/can/listener.py#L18) is really not doing much.

felixdivo, 2018-04-08:
Closing this for now. Please reopen if the problem still exists.

closed 2018-04-08 by felixdivo