hardbyte/python-can #252: bitrate settings error of a Vector Bus

matisla, 2018-02-16:
When I create an instance of Bus like this:

`can.interface.Bus(bustype='vector', channel=0, bitrate=125000)`

I get: `Can not set bitrate since no init access`
but without the bitrate parameter, I don't get the error.

I would have two different bitrate on each channel of my CANCaseXL,
but i'm not sure if it's work correctly without this parameter.

Could someone explain to me why this error occure and say if it's a bug or not ?

christiansandberg, 2018-02-17:
Do you have another application connecting to the CAN bus at the same time? That could prevent python-can from changing the bitrate. This is normal for other interfaces so I think that making this a warning was a little harsh.

Also we don’t support setting separate bitrates yet.

matisla, 2018-02-17:
Ok, thanks for the explanation !
 I hope it will possible to have multiple bitrates.

closed 2018-02-17 by matisla