hardbyte/python-can #260: Add testing under Windows

felixdivo, 2018-02-20:
Currently our code is automatically tested under Linux and many different Python versions, and under OSX under only two Python 3 versions using Travis CI. (see `python-can/.travis.yml`) 

But no automatic testing under Windows is performed. Should we integrate it into the testing of python-can? Unfortunately Travis CI can't do that, but maybe someone else knows a similar service for Windows.

christiansandberg, 2018-02-21:
Unfortunately the unit tests we run have very little coverage of the whole project, so testing it on Windows as well won't help much.

What we really need is a physical setup with all the interfaces we support so that we could test them all against each other. Maybe we could ask the different vendors to give us free samples... :)

felixdivo, 2018-02-21:
Hm yes, maybe you are right. But then we would still need someone running a CI server, something like [buildbot](https://docs.buildbot.net/current/tutorial/firstrun.html) maybe.

hardbyte, 2018-02-21:
Let's keep this issue just to basic testing support on windows :-) I've created issue https://github.com/hardbyte/python-can/issues/261 for the test rig.

https://www.appveyor.com/ is the closest I've heard of to travis but for windows

felixdivo, 2018-02-22:
Yeah, that looks nice. I've seen it being used in some other Python projects as well. Could you set it up?

hardbyte, 2018-02-22:
Happy to make the project now - do you wan to take on creating a branch with a  `.appveyer.yml` config?

https://ci.appveyor.com/project/hardbyte/python-can

felixdivo, 2018-02-22:
Yeah, I can have a look at it.

PR #264 (2018-02-23): Add CI testing on Appveyor (Windows)
This adds CI tests on Windows. I also factored out common skipping behavior (`@skipIf(IS_TRAVIS, "...")`) and replaced it with generic CI skipping behavior (`@skipIf(IS_CI, "...")`). The constants are defined in `test/config.py` and define some properties in a central place:

- IS_TRAVIS
- IS_APPVEYOR
- IS_CI
- TEST_CAN_FD
- More? Maybe whether to test on hardware, but is that the right place for stuff like that? Your thoughts?

Closes #260.

Still work in progress.
closed 2018-02-23 by felixdivo