hardbyte/python-can #262: Documentation improvements

felixdivo, 2018-02-22:
Running `sphinx-build doc/ doc/_build -n` reveals many apparent problems. Do we need to solve them?

hardbyte, 2018-02-23:
Yes as it probably effects how the docs are rendered. I'll take a look.

felixdivo, 2018-03-16:
Maybe we should test this on the CI server?

felixdivo, 2018-08-06:
Update on this: The docs are now built as part of the Python 3.6 test on Travis CI with the option `-n` to reveal any problems. They are not considered errors for now. Any help is appreciated.

hardbyte, 2019-02-23:
Here are the errors in master from the 3.1 merge:

```
/home/travis/build/hardbyte/python-can/can/interface.py:docstring of can.Bus:1: WARNING: py:class reference target not found: can.bus.BusABC
/home/travis/build/hardbyte/python-can/can/bus.py:docstring of can.BusABC.send_periodic:9: WARNING: py:meth reference target not found: CyclicTask.stop
/home/travis/build/hardbyte/python-can/can/bus.py:docstring of can.BusABC.send_periodic:22: WARNING: py:meth reference target not found: stop
/home/travis/build/hardbyte/python-can/can/bus.py:docstring of can.BusABC.state:: WARNING: py:class reference target not found: NamedTuple
/home/travis/build/hardbyte/python-can/can/thread_safe_bus.py:docstring of can.ThreadSafeBus:1: WARNING: py:class reference target not found: ObjectProxy
/home/travis/build/hardbyte/python-can/can/thread_safe_bus.py:docstring of can.ThreadSafeBus:10: WARNING: py:meth reference target not found: can.BusABC._recv_internal
/home/travis/build/hardbyte/python-can/can/thread_safe_bus.py:docstring of can.ThreadSafeBus:10: WARNING: py:meth reference target not found: can.BusABC._recv_internal
/home/travis/build/hardbyte/python-can/can/interfaces/canalystii.py:docstring of can.interfaces.canalystii.CANalystIIBus:1: WARNING: py:class reference target not found: can.bus.BusABC
/home/travis/build/hardbyte/python-can/can/interfaces/iscan.py:docstring of can.interfaces.iscan.IscanBus:1: WARNING: py:class reference target not found: can.bus.BusABC
/home/travis/build/hardbyte/python-can/doc/interfaces/ixxat.rst:11: WARNING: py:meth reference target not found: can.interfaces.ixxat.canlib.IXXATBus.send_periodic
/home/travis/build/hardbyte/python-can/can/interfaces/kvaser/canlib.py:docstring of can.interfaces.kvaser.canlib.KvaserBus:1: WARNING: py:class reference target not found: can.bus.BusABC
/home/travis/build/hardbyte/python-can/can/interfaces/kvaser/canlib.py:docstring of can.interfaces.kvaser.canlib.KvaserBus.get_stats:: WARNING: py:class reference target not found: can.interfaces.kvaser.structures.BusStatistics
/home/travis/build/hardbyte/python-can/can/interfaces/ics_neovi/neovi_bus.py:docstring of can.interfaces.ics_neovi.NeoViBus:1: WARNING: py:class reference target not found: can.bus.BusABC
/home/travis/build/hardbyte/python-can/can/interfaces/nican.py:docstring of can.interfaces.nican.NicanBus:1: WARNING: py:class reference target not found: can.bus.BusABC
/home/travis/build/hardbyte/python-can/can/interfaces/nican.py:docstring of can.interfaces.nican.NicanBus:5: WARNING: py:meth reference target not found: can.interfaces.nican.NicanBus.__init__
/home/travis/build/hardbyte/python-can/can/interfaces/pcan/pcan.py:docstring of can.interfaces.pcan.PcanBus:1: WARNING: py:class reference target not found: can.bus.BusABC
/home/travis/build/hardbyte/python-can/can/interfaces/pcan/pcan.py:docstring of can.interfaces.pcan.PcanBus:3: WARNING: py:meth reference target not found: can.interface.pcan.PcanBus.flash
/home/travis/build/hardbyte/python-can/can/interfaces/pcan/pcan.py:docstring of can.interfaces.pcan.PcanBus:3: WARNING: py:meth reference target not found: can.interface.pcan.PcanBus.status
/home/travis/build/hardbyte/python-can/can/interfaces/pcan/pcan.py:docstring of can.interfaces.pcan.PcanBus:: WARNING: py:class reference target not found: can.bus.BusState
/home/travis/build/hardbyte/python-can/can/interfaces/pcan/pcan.py:docstring of can.interfaces.pcan.PcanBus.state:: WARNING: py:class reference target not found: NamedTuple
/home/travis/build/hardbyte/python-can/can/interfaces/serial/serial_can.py:docstring of can.interfaces.serial.serial_can.SerialBus:1: WARNING: py:class reference target not found: can.bus.BusABC
/home/travis/build/hardbyte/python-can/can/interfaces/serial/serial_can.py:docstring of can.interfaces.serial.serial_can.SerialBus:3: WARNING: py:meth reference target not found: can.interfaces.serial.SerialBus._recv_internal
/home/travis/build/hardbyte/python-can/can/interfaces/slcan.py:docstring of can.interfaces.slcan.slcanBus:1: WARNING: py:class reference target not found: can.bus.BusABC
/home/travis/build/hardbyte/python-can/can/interfaces/socketcan/socketcan.py:docstring of can.interfaces.socketcan.CyclicSendTask:1: WARNING: py:class reference target not found: can.broadcastmanager.ModifiableCyclicTaskABC
/home/travis/build/hardbyte/python-can/can/interfaces/socketcan/socketcan.py:docstring of can.interfaces.socketcan.CyclicSendTask:1: WARNING: py:class reference target not found: can.broadcastmanager.RestartableCyclicTaskABC
/home/travis/build/hardbyte/python-can/can/interfaces/socketcan/socketcan.py:docstring of can.interfaces.socketcan.SocketcanBus:1: WARNING: py:class reference target not found: can.bus.BusABC
/home/travis/build/hardbyte/python-can/can/interfaces/socketcan/socketcan.py:docstring of can.interfaces.socketcan.SocketcanBus:1: WARNING: py:meth reference target not found: can.BusABC._detect_available_configs
/home/travis/build/hardbyte/python-can/can/interfaces/systec/ucanbus.py:docstring of can.interfaces.systec.ucanbus.UcanBus:1: WARNING: py:class reference target not found: can.bus.BusABC
/home/travis/build/hardbyte/python-can/can/interfaces/systec/ucanbus.py:docstring of can.interfaces.systec.ucanbus.UcanBus:: WARNING: py:class reference target not found: can.bus.BusState
[... cut ...]
/home/travis/build/hardbyte/python-can/can/io/sqlite.py:docstring of can.SqliteReader.read_all:: WARNING: py:class reference target not found: Generator
/home/travis/build/hardbyte/python-can/can/io/asc.py:docstring of can.ASCWriter:1: WARNING: py:class reference target not found: can.listener.Listener
/home/travis/build/hardbyte/python-can/can/io/canutils.py:docstring of can.CanutilsLogWriter:1: WARNING: py:class reference target not found: can.listener.Listener
/home/travis/build/hardbyte/python-can/can/io/blf.py:docstring of can.BLFWriter:1: WARNING: py:class reference target not found: can.listener.Listener
```

felixdivo, 2019-02-24:
Replacing `.. autoclass:: can.Listener` with `.. autoclass:: can.listener.Listener` in `listeners.rst` solves many `WARNING: py:class reference target not found: can.listener.Listener` warnings. But it obviously shows the class as `class can.listener.Listener:`, which is not what we want. How can we solve this?

felixdivo, 2019-02-24:
I was able to solve

```
[...]/python-can/can/interfaces/pcan/pcan.py:docstring of can.interfaces.pcan.PcanBus.state:: WARNING: py:class reference target not found: collections.namedtuple
```

and opened an issue in the sphinx repo since it seems to be a problem in the docs. sphinx-doc/sphinx#6100

felixdivo, 2019-02-26:
> Replacing `.. autoclass:: can.Listener` with `.. autoclass:: can.listener.Listener` in `listeners.rst` solves many `WARNING: py:class reference target not found: can.listener.Listener` warnings. But [...]

This is also a problem with `BusABC` and the `can.broadcastmanager.*` module. If we can fix this, we should be able to shrink the number of issues by a lot. I could not find any helpful advice yet ...

felixdivo, 2021-04-14:
Maybe these could help (added in [Sphinx Release 4.0.0 beta1 (released Apr 12, 2021)](https://github.com/sphinx-doc/sphinx/blob/4.x/CHANGES#L122)):

- https://www.sphinx-doc.org/en/master/usage/restructuredtext/domains.html#directive-option-py-function-canonical
- (https://www.sphinx-doc.org/en/master/usage/configuration.html#confval-python_use_unqualified_type_names)

felixdivo, 2022-11-16:
@zariiii9003 Is this still relevant?

hardbyte, 2022-11-24:
I think we can close it, we've made significant progress improving the docs and reducing the warnings

closed 2022-11-24 by hardbyte