hardbyte/python-can #271: python-can logo

hardbyte, 2018-02-24:
What are everyones thoughts on having a logo for python-can?

Add a :+1: or :-1: or a new idea...

hardbyte, 2018-02-24:
**Entry 1**
White on black, inspiration from micropython:

[image]

hardbyte, 2018-02-24:
**Entry 2**

Project initials in a circuit layout:

[image]

felixdivo, 2018-02-24:
Well, this issue was unexpected. ^^

felixdivo, 2018-02-24:
I actually like the twisted wires from this one:

[image]

cowo78, 2018-02-28:
Something like what felix proposes but with the penguin substituted by i.e. https://pythonhow.com/cropped-python-logo-python-how-png/ since the package is not linux specific?

hardbyte, 2018-07-01:
That could work - although we'd need to check with the PSF if we can use their trademarked logo.

Here is a variant:

[image]

hardbyte, 2018-07-01:
Here is another variant:

[image]

felixdivo, 2018-07-01:
@hardbyte From the [PSF Trademark Usage Policy](https://www.python.org/psf/trademarks/) (section _General Goals_):

> We do not want these trademarks to be used [...] in a way that is misleading or may imply association of unrelated modules, tools, documentation, or other resources with the Python programming language

hardbyte, 2018-07-03:
Square variant:

[image]

bhass1, 2019-04-02:
How about the square variant and longer variant with both Python snakes?? (Add head to the blue one)

bhass1, 2019-05-03:
**TOP-TOP Python CAN-bus:**
[image]

**TOP-BOTTOM Python CAN-bus:**
[image]

felixdivo, 2022-11-16:
@hardbyte Is this also a discussion?