hardbyte/python-can #272: Do you use python-can? Tell us!

hardbyte, 2018-02-24:
This is not an issue so much as a lightweight way of gathering information on who is using python-can. This is mostly to satisfy our curiosity, but might also help us decide how to evolve the project.

So, if you use python-can, please post here and tell us more!

felixdivo, 2018-02-24:
I use python-can as part of a student project at TU Darmstadt, Germany that I work on with many others. We are the [Sailing Team Darmstadt](http://www.st-darmstadt.de) and our goal is to develop a robotic sailing ship that is capable of crossing the Atlantik fully autonomously. And as it uses a CAN bus internally, we needed a library for than, and python-can was there to be used. And as reliability is a huge topic with that project I am very interested in testing the library as thoroughly as possible, which explains many of my efforts to enable better testing in here.

christiansandberg, 2018-02-25:
At the company I used to work for, which develop electric motor drives, we started using Python in various testing and automation tasks. Kvaser is the most common interface used but I counted five different interfaces used in different parts of the company and contexts. So python-can was really helpful when writing scripts and have them executed anywhere.

I also open sourced a CANopen library built on top of python-can.

cowo78, 2018-02-28:
Worked for a company that builds net weight filling devices (packaging area). Used python-can to setup and test custom electronics. Setup meant in our case remote programming and testing was a test protocol that involved (simple) custom hardware. All of this using Windows and IXXAT devices.

For those that might be interested in the area, the interesting project https://github.com/arachnidlabs/cyflash uses python-can to remotely reflash firmware on a range of microcontrollers.

inneralien, 2018-03-01:
Currently using python-can for industrial machine telemetry.

TravisJoe, 2018-03-01:
I have used it on several products, nothing in large scale production, but fairly significant amounts of work.

In one instance I did write a more specific native python socket-can that only had the minimum of what I needed. But if it was not for Python-CAN I would not of been able to get started quickly and reliably. I have highly recommended it to others.

anxuae, 2018-03-07:
I use `python-can` at work for automated testing purpose on real infotainment ECU that my company is developing. We mainly use it on Linux with ` 	socketcan_ctypes` to validate functional signals from CAN to the OS responsible for HMI management.
The next job that we have started is the development of a DoIP stack in Python (it may be a new Bus interface integrated using `pluggy` ?).
We have also the ambition to manage LIN networks.

mgiaco, 2018-03-15:
I use python-can at work. We made some different implementations for our in house CAN protocol. Python CAN helps us a lot because we must support different USB to CAN devices. And I also recommended it to other friends.

edobez, 2018-03-16:
I'm using `python-can` at work with a PCAN-USB device. We are testing a medical device that we produce, as a black box.
Used with https://github.com/construct/construct, it's really simple to build frames from higher-level structures, like dictionaries.

Bobet21, 2018-05-28:
Hey, I am using Pyton-can to understand how my car is behaving, looking for reverse engineering tools, can filtering and live plotting.

mitchdetailed, 2018-06-07:
Yes, I use it to reverse engineer my vehicle's canbus, and create a csv file for which i can use in a data analysis program called 'Megalog Viewer HD' https://youtu.be/tNAmKgHUyyw

gustavovelascoh, 2018-08-08:
Hi, I just started with CAN/CANOpen and I'm using python-can/canopen to test a new encoder. This helped me to set up quickly a test environment and I'm looking forward to use more devices.

milhead2, 2018-08-09:
I've been using python-can for a couple years.  

At my company we develop gauges for on and off-road vehicles.  

Running on top of python-can we use robotframework and python-j1939 to run multiple concurrent CAN controller simulations which emulate vehicle bus conditions and ambient bus traffic.  All of our new features are exercised and regression tested with robotframework scripts.  In this environment we can develop/test gauge features at the J1939 level.

We maintain the python-j1939 module but don't have too many users outside ourselves as it's a little dated and generally adopters have commercial tools.

hardbyte, 2018-09-23:
https://github.com/CaringCaribou/caringcaribou

rukie, 2018-12-18:
I use python-can along with cantools to control various CAN operated hardware on vehicles, during development. Significantly more cost effective than Control Desk, Simulink, etc. J1939 is everywhere here.

mindfuucker, 2018-12-19:
We create stationary energy storage solutions for grid optimization and and emergency power. Our batteries are from second-use EVs. We use the library to extract the (proprietary) data for further use.

lcvisser, 2019-03-08:
I wanted to, but I have an IXXAT USB-to-CAN device, which is not supported under Linux. I developed some code on top of ECI that maybe I can share (I'd need to check with management). Let me know if there's interest.

rusoku, 2019-03-10:
Today my wife's NISSAN says first "hello world" using Python-CAN ;-)

bhass1, 2019-03-29:
Use at work for various internal tools; used at school for CAN bus fuzzing. The common API for various CAN-USB dongles is very appealing.

stravs, 2019-07-04:
i use it at work in a prototype electric propulsion system for boats. Thank (all of) you for making my life easier! :)

mcdonl1, 2019-08-09:
Using it to build a remote vehicle monitoring system for heavy machinery.

andriy-neborak, 2019-10-10:
I just begin use python-can with IXXAT USB-CAN Compact module. I write project for device testing via CAN bus. 
Pros for me are:
1. supports several well-known modules supported
2. simplify connects to python based web frameworks.

divyamSoni, 2020-05-13:
I'm using python-can (with PySide2) to make a desktop application to diagnose Electric Vehicle Batteries via CAN-bus

mcdonl1, 2020-05-13:
Hello

divyamSoni, 2020-05-13:
> Hello

Hey!

SoloITBiz, 2020-07-16:
I am using python-can to gather vehicle telemetry for Darwin Innovation Group's continuous connected vehicles.
Later on there is the intention to interact with the vehicle control systems over the canbus.

DustinMoriarty, 2020-10-22:
We use it for hardware in the loop testing.

pavel-kirienko, 2021-05-08:
We use it in [PyUAVCAN](https://github.com/UAVCAN/pyuavcan/) -- a Python implementation of the [UAVCAN](https://uavcan.org) protocol stack.

tjohanssonn, 2021-06-18:
I typically use it for various hobby projects or rapid prototyping at work. I love this library, especially the portability aspect that enables me to easily run it on lots of different hardware. 

> I'm using python-can (with PySide2) to make a desktop application to diagnose Electric Vehicle Batteries via CAN-bus

Interesting! I'm doing a project where I try to combine these. How do you pass the information from python-can to PySide?