hardbyte/python-can #273: Testing the SocketCAN backend on Travis CI

felixdivo, 2018-02-24:
Is it possible to test the SocketCAN backend on Travis CI using the virtual can channels provided by the kernel driver like `vcan0` or `vcan1`?

hardbyte, 2018-02-25:
I doubt it - just need to see if you can run `modprobe vcan` and the various `ip` commands required.

felixdivo, 2018-02-28:
Actually it works! See [here](https://travis-ci.org/hardbyte/python-can/jobs/347319036#L451).

felixdivo, 2018-03-02:
I submitted a proof-of-concept in #275.

felixdivo, 2018-06-14:
This should probably be done in a more general bus test cleanup. I am currently waiting for #290 to be merged. The idea ist to write a single test case with many tests, and then apply that test to all the interfaces, possibly adding dome more specific tests and opting in/out of special/unsupported bahaviour tests.

hardbyte, 2018-07-14:
I see the travis build of the `develop` branch is broken due to this?

https://travis-ci.org/hardbyte/python-can/jobs/403679419

christiansandberg, 2018-07-14:
Yes, I was surprised that we had `sudo: false` but used sudo anyway and it worked, but I guess it just worked temporarily.

felixdivo, 2018-07-16:
[It now works](https://travis-ci.org/hardbyte/python-can/builds/404328656). The build times didn't increase too much as well.

felixdivo, 2019-02-10:
This is done in `test/back2back_test.py`.

closed 2019-02-10 by felixdivo