hardbyte/python-can #280: Specify bitrate in a config file?

randycoulman, 2018-03-14:
Is there a way to specify the `bitrate` parameter in a configuration file?  From what I can see, there isn't.

I'd like to be able to supply a config file with parameters that work on a CI machine but allow a developer to override those values using environment variables when working on their local workstation.  There is already an environment variable for this, but the current code doesn't seem to use a `bitrate` that was provided in a configuration file.

I saw #172, but that PR doesn't seem to show any actual changes and it was closed without merging after being approved.

Is there a reason why this wouldn't be a desirable feature?

Thanks for a super-handy library!  Works great in all of my testing so far!

felixdivo, 2018-03-15:
Maybe it should simply pass all arguments to the bus constructor, so you can also use some other interface specific parameters.

christiansandberg, 2018-03-15:
This _should_ work. Everything in the config file will be read into the configuration as strings:
https://github.com/hardbyte/python-can/blob/develop/can/util.py#L73

Then the bitrate will be converted to an integer:
https://github.com/hardbyte/python-can/blob/develop/can/util.py#L164

christiansandberg, 2018-03-15:
No, it seems broken. In can/interface.py, only interface and channel are propagated to **kwargs.

randycoulman, 2018-03-15:
I'm wondering if something like this would work:

- In `Bus.__new__`, include all of the `kwargs` as part of the config being passed to `load_config` (https://github.com/hardbyte/python-can/blob/develop/can/interface.py#L64-L67)

- Pass the resulting config through to the backend constructor (https://github.com/hardbyte/python-can/blob/develop/can/interface.py#L96)

There'd have to be some other changes as well, like deleting the `bustype` field before calling `load_config` and removing the `interface` and `channel` entries from the resulting config to handle them as they are now.

But the overall result should be that the config file can provide any valid parameters (including `bitrate`) for a particular interface and they will make it through to the constructor call.

felixdivo, 2018-03-15:
I just wanted to note that I changes some of that config handling in PR #278, specifically [here](https://github.com/hardbyte/python-can/pull/278/files#diff-3194f59bb641265893300991077dcd9dR111).

randycoulman, 2018-05-16:
@felixdivo Sorry, I'm not on that project any more (it was a consulting client) and I don't have access to the codebase to check anything out.

PR #309 (2018-05-28): cleanups,fixes & docs in Bus.__new__() and util.load_config()
This PR changes `util.load_config()` to pass through any unused parameters. It is made more clear in the code what `Bus.__new__()` is doing and what not. A lot has moved to `util.load_config()` to remove code and responsibility duplication. The documentation has each been changed to reflect what the methods are actually doing, that was not done correct before. 

Based an @randycoulman's comment [over here](https://github.com/hardbyte/python-can/issues/280#issuecomment-373426370)
Fixes #280
closed 2018-05-28 by felixdivo