hardbyte/python-can #282: Can not send CAN FD frame

skpang, 2018-03-18:
I'm trying to send a CAN FD frame using socketcan. I've tested it first using cansend.

cansend can0 7df##15555555555555555

This is working fine as seen on a PCAN-USB Pro FD analyser. Then I tried using python3.4
I get an error when I set the is_fd =1

This is screen printout.

Python 3.4.4 (default, Apr 17 2016, 16:02:33) 
[GCC 5.3.1 20160409] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import can
>>> bus = can.interface.Bus(channel='can0', bustype='socketcan_native')
>>> msg = can.Message(arbitration_id=0x7de,is_fd = 0, data=[0, 25, 0, 1, 3, 1, 4, 1])
>>> print(msg)
Timestamp:        0.000000    ID: 000007de    X          DLC: 8    00 19 00 01 03 01 04 01
>>> bus.send(msg)
>>> msg = can.Message(arbitration_id=0x7de,is_fd = 1, data=[0, 25, 0, 1, 3, 1, 4, 1])
>>> print(msg)
Timestamp:        0.000000    ID: 000007de    X     F    DLC: 8    00 19 00 01 03 01 04 01
>>> bus.send(msg)
Traceback (most recent call last):
  File "/home/pi/python-can-develop/can/interfaces/socketcan/socketcan_native.py", line 475, in send
    bytes_sent = self.socket.send(build_can_frame(msg))
OSError: [Errno 22] Invalid argument

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/pi/python-can-develop/can/interfaces/socketcan/socketcan_native.py", line 477, in send
    raise can.CanError("Transmit failed (%s)" % exc)
can.CanError: Transmit failed ([Errno 22] Invalid argument)

 Anyone know how this can be fixed ?

christiansandberg, 2018-03-18:
You also need to specify that CAN FD should be supported when creating the bus:

bus = can.interface.Bus(channel='can0', bustype='socketcan_native', fd=True)

skpang, 2018-03-18:
Yes, that works.

Thanks

closed 2018-03-18 by skpang