hardbyte/python-can #287: Building an executable with Pyinstaller

ChrisAlphabet, 2018-03-27:
Hi all,

I am trying to build an executable for a Python project that uses the Kvaser interface.

Running `pyinstaller canbus_exerciser.py` works fine and creates the executable.
But when I run the executable from a command prompt I get:

```python
ImportError: No module named 'can.interfaces.kvaser
```

which seems to suggest that the executable can't find the device driver for the Kvaser.

Using `python canbus_exerciser.py` works perfectly so I know that the driver has been installed properly.

A quick look at the pyinstaller docs shows that you can pass in additional binary and non-binary files.

There are a bunch of files in `C:\Program Files\Kvaser\Drivers` but I am not sure which ones need to be used with pyinstaller to make the Kvaser device driver work properly.

Has anyone tried to do anything similar and know which files to select?

Thanks!

christiansandberg, 2018-03-29:
The problem is that the interface modules are not imported using the `import` statements which is why PyInstaller doesn't find them. You will need to tell PyInstaller explicitly that can.interfaces.kvaser should be included too.

ChrisAlphabet, 2018-04-04:
I tried adding 'can.interfaces.kvaser' to the hidden imports part of the `.spec` file but I get the error:  

```
Import Error: Cannont import module can.interfaces.kvaser for CAN 
interface 'kvaser' : 'LibraryLoader' object is not callable 
```

Is that the right place to specify an import?

ChrisAlphabet, 2018-04-04:
That is the right place to add it! :smile: 

I did a clean build using `pyinstaller --clean canbus_exerciser.spec` and everything is working.

Thanks for your help! :+1:

closed 2018-04-04 by ChrisAlphabet

gauravagnihotri, 2019-03-05:
Man, i spent hours trying to figure this out. Thanks a lot for the solution!

pradeeprajalingam, 2021-11-19:
Can you please let me know how to add 'can.interfaces.kvaser' to the hidden imports part of the .spec file?

JRYman92, 2024-11-11:
Can you please let me know how deal it

EnzoBurille, 2025-06-03:
> Can you please let me know how deal it

pyinstaller --hidden-import=can.interfaces.kvaser your_file.py