hardbyte/python-can #294: Got "XL_ERR_INVALID_PORT" on Vector - CANCaseXL

matisla, 2018-04-10:
When I try to use create a can bus like:
`Bus(channel=0, bustype='vector' app_name='CANalyzer', bitrate=250000)`

I get an exception: `"XL_ERR_INVALID_PORT"`
I looked in the Vector documentation, but I only found that the error correspond to:
"The driver attempted to access a port by an invalid pointer or index."

I checked my configuration with Vector Hardware, 
I also be able to use the channel with CANalyzer.

Have someone an Idea where is the problem ?

mgiaco, 2018-04-10:
I tested the lates release for you.. I use the same settings with python 3.5

Bus(bustype='vector', app_name='CANalyzer', channel=0, bitrate=250000)

works perfect. Have you tried some other tool with the device? Canalyzer or Busmaster to check if everything is okay with the hardware?

matisla, 2018-04-10:
after rewriting partially my code (to find the bug), I realize that the problem came from using multiprocessing lib (so an other Process) and the reference of the bus was not shared ....

Thanks for your quick response.

closed 2018-04-13 by christiansandberg