hardbyte/python-can #297: NeoVi transmit and receive (ICS)

ltatgithub, 2018-04-13:
i am trying to use the example given to send and receive the CAN frames with NeoVi fire 2. I am receiving some junk like this () and no sent frames received other side. but the same code works fine for PCan. 
`import time

import can
import ics 

if __name__ == '__main__':
    devices = ics.find_devices()

    for device in devices:
        print(device.Name, device.SerialNumber)
    
    bus = can.interface.Bus(bustype='neovi',
                            channel=42,
                            receive_own_messages=False,bitrate=1000000)

    bus.send(can.Message(arbitration_id=1, extended_id=True))
    bus.send(can.Message(arbitration_id=2, extended_id=True))
    bus.send(can.Message(arbitration_id=3, data=[0, 25, 0, 1, 3, 1, 4, 1],extended_id=False))
    time.sleep(1)
    a = True
    b = 0
    

    while b<100:
        m = bus.recv(0.001)
        if m != None:
            print 'msg=',m.data
            print 'ID = ',m.arbitration_id
            b +=1
            print 'b =', b
    print "shutdown"
    bus.shutdown`

I am using PCan to view and send codes to python connected to NeoVi fire 2

ltatgithub, 2018-04-27:
Hello @pierreluctg,
thank you for replying.
same printing method is used for other devices like PCAN and vector Can. I can see proper data printing.
 no other device is sending on this CAN bus. same device Neovi which I am using is sending like this
`bus.send(can.Message(arbitration_id=3, data=[0, 25, 0, 1, 3, 1, 4, 1],extended_id=False))` as above program

I am receiving something like this 

_(u'neoVI FIRE2', 782842544)
msg= �
ID =  0
b = 1
msg= �
ID =  0
b = 2
msg= �
ID =  0
b = 3
msg= �
ID =  0
b = 4
msg= �
ID =  0
b = 5
msg=  ��
ID = 3
b = 6_

when i change ID, I can see printed ID matches. but again data is not readable.

pierreluctg, 2018-04-27:
if you run `python -m can.logger -i neovi -c 42` in a console when running your code, what do you get?

Note that you may have to enable the 'server mode' in neoVI Explorer to allow more that one client.

ltatgithub, 2018-05-22:
@pierreluctg i was not able to read that because NeoVi fire 2 doesn't have the python lib. we need to write a wrapper with ICS lib and use.

closed 2018-05-22 by ltatgithub