hardbyte/python-can #299: Unable to set filters for Vector CancaseXL

Alvin80, 2018-04-17:
I've tried to set up filters without any result. 
Below my environment
OS: Windows7
Python version: 3.6.5
HW CAN: CANcaseXL
CANcase Driver: 8.7.16

Below my code snippet
```
bus = can.interface.Bus(bustype='vector', app_name='CANalyzer', can_filters=[{"can_id": 0x67C, "can_mask": 0xFFF, "extended": False}], channel=0, bitrate=125000, rx_queue_size=10000, receive_own_messages=False)

notifier = can.Notifier(bus, [can.Printer()])
```

I don't receive any error but i'm not able to filter the packets: all can messages are printed by the notifier.
How can i fix this problem?

felixdivo, 2018-04-17:
Maybe this could be fixed with #277?

christiansandberg, 2018-04-18:
I've tested the filtering with a virtual channel and that seems to work. It could be that the CANcaseXL doesn't support hardware filtering.

Alvin80, 2018-04-18:
How can I check if hardware filtering is supported by CANcaseXL? Any suggestions?

felixdivo, 2018-06-04:
No, I cannot help you with that. But #277 added quite some changes to the internal filtering. As you can see [in the code](https://github.com/hardbyte/python-can/blob/develop/can/interfaces/vector/canlib.py#L178), the bus should now fall back to software filtering if setting the filters was no successful. That should solve you problem. You can test it by installing the version on the `develop` branch.

closed 2018-06-04 by felixdivo