hardbyte/python-can #301: Release the interface after it's used?

wangdino, 2018-04-27:
Hi, first of all thanks for the lib!

I wonder if there is a way to release the CAN interface if I finished running my program. When I try to run the same code multiple times I'll end up having multiple instances of interface being used by the same application. Taking a Vector virtual CAN as an example:
[image]

Thanks for any insights!

christiansandberg, 2018-04-27:
Do you use the .shutdown() method?

felixdivo, 2018-04-27:
As far as I know, `bus.shutdown()` closes the connection that is tied to that instance of the bus class. If you have multiple connections, you must've had multiple instances. You need to call `shutdown()` on each of them.

wangdino, 2018-04-27:
Thanks I think for me it's probably easier to keep track if I have a `bus.shutdown()` for each instance in the code so they all eventually get released.

Issue closed.

closed 2018-04-27 by wangdino