hardbyte/python-can #305: Callback functionality for periodic send tasks

zorce, 2018-05-04:
Some way to provide a callback och generators for the frame data in a periodic message.
The use case I have is to update a checksum and a rolling counter for each frame.

Or should this be implemented using timers and a raw socket with manual send after calculations?

christiansandberg, 2018-05-04:
The problem is that callbacks won't work when periodic transmission is done in hardware or kernel. The idea is that the interface should be the same, but it can be discussed.

zorce, 2018-05-04:
I just came to the same conclusion, so for now I have implemented it with python timer class. Not sure yet if this will work fast enough for 10ms frames.

felixdivo, 2018-06-12:
Implementing it with a timer sounds good. If timing really is a big problem here, I doubt that a callback would be significantly faster. Has it worked with a timer?

zorce, 2018-06-13:
It seems to work quite good, have only tested with the vcan bus so far but it seams to keep up quite good.
Will do more testing with real bus and measure it using CANoe soon.

felixdivo, 2018-06-13:
Sounds good! If any problems occur, please reopen the ticket and we'll have a further look at it.

closed 2018-06-13 by felixdivo