hardbyte/python-can #306: Listener for sent messages

teebr, 2018-05-10:
I'm using a PCAN-USB adaptor with a Windows 10 machine. I'd like to log all messages in either direction. For received messages I'm using a standard listener (`ASCWriter`), but I can't see a simple way of doing this for outgoing messages. 

I've seen a couple of devices (e.g. Vector) have a `receive_own_messages` flag, but not PCAN. It doesn't seem like I can set it to loopback mode on Windows, though I may well be wrong?

As a temporary solution, I've subclassed `PCanBus` and added the listeners I'm using to receive messages as an attribute. Then I've overwritten the `send()` method, copying exactly the source code but then adding a couple of lines to call the `on_message_received()` method for each listener, with the message which was just sent. Obviously this isn't a great long-term solution.

If anyone has any advice on loopback etc for the PCAN then I'm fine with that solution, otherwise is there an easier means of logging sent messages via the library?

felixdivo, 2018-05-22:
Well, a software-only solution might be the simplest, although it should be handled only when `recv()` is called. I don't know whether PCAN-USB even supports a loop-back mode.

christiansandberg, 2018-05-22:
I don’t think it does. The problem is getting the timestamp exactly right for when the message actually went out on the bus, so therefore it must be done by the driver.

felixdivo, 2018-05-23:
Well that would surely be better if it is possible. But I'm not sure about it being supported by the API. PCAN seems to be a bling spot concerning documentation anyways, see #56.

teebr, 2018-05-23:
@christiansandberg - yes, that's why [PCAN doesn't offer timestamping of sent messages](https://www.peak-system.com/forum/viewtopic.php?f=7&t=2035). I couldn't find any time function in Python which exactly aligned with the PCAN timestamp (from memory `uptime.uptime()` and some others were a few seconds off). At the moment I'm using the timestamp of a received message as a reference point, and from then on using `time.perf_counter()` to get a relative time. So far I haven't had any issues with this but my concern is that for longer periods it may go slightly out of sync with the PCAN timestamp.

felixdivo, 2018-09-07:
Well, you could resync it with the first message that comes after each 5 minutes or something.

felixdivo, 2019-09-04:
This will not be implemented because it's hard to do precisely, but users would assume it it. If someone definitely requires it, one might send each message onto a virtual bus as well, where all interested listeners can attach. Or simply add some custom send method, which basically sends it normally and then calls some callback afterwards.

closed 2019-09-04 by felixdivo