hardbyte/python-can #308: Is 'socketcan_native' bus supported in python 2.7?

gimohsent, 2018-05-15:
Hello. Thanks for the awesome work.
The following script works executes with python 3 interpreter.

```
import can
import time
import os

os.system('sudo /sbin/ip link set can0 up type can bitrate 500000')

bus = can.interface.Bus(channel='can0', bustype='socketcan_native')
msg = can.Message(arbitration_id=0x7de,data=[0x00,0x01,0x02,0x03,0x04,0x05,0x06,0x07], extended_id=False)

while True: 
      bus.send(msg)
      time.sleep(1)
```

 But using python 2.7 results in error:
```

No handlers could be found for logger "can.socketcan.native"
Traceback (most recent call last):
  File "/home/quad/Desktop/can_test.py", line 12, in <module>
    bus = can.interface.Bus(channel='can0', bustype='socketcan_native')
  File "/usr/local/lib/python2.7/dist-packages/python_can-1.4.1-py2.7.egg/can/interfaces/interface.py", line 38, in __new__
    from can.interfaces.socketcan_native import SocketscanNative_Bus
  File "/usr/local/lib/python2.7/dist-packages/python_can-1.4.1-py2.7.egg/can/interfaces/socketcan_native.py", line 23, in <module>
    raise ImportError()
ImportError
```

christiansandberg, 2018-05-15:
No, native is not supported on 2.7. Just use 'socketcan' instead of 'socketcan_native' and it will choose between native and ctypes depending on Python version.

gimohsent, 2018-05-16:
That fixed it. Thanks!

closed 2018-05-16 by mohse-n