hardbyte/python-can #310: can.send : No buffer space available

odenizot, 2018-05-17:
Hello everybody,

I use a Raspberry pi 3B+ with PiCan2 Duo board, python 3.5.3 and raspbian stretch (with desktop, 2018-04-18, kernel 4.14).
The deal is to receive a frame from can0 and just sent it to port can1.
Here is the code : 

```
import can

can0 = can.interface.Bus(channel = 'can0', bustype = 'socketcan_native')
can1 = can.interface.Bus(channel = 'can1', bustype = 'socketcan_native')

while True:
  msg = can0.recv()
  can1.send(msg)
  print('.',end='')
```

I receive the `No buffer space available error`, only 11 frames can be sended : 

```
...........Traceback (most recent call last):
File "/usr/local/lib/python3.5/dist-packages/can/interfaces/socketcan/socketcan_native.py", line 474, in send
    bytes_sent = self.socket.send(build_can_frame(msg))
OSError: [Errno 105] No buffer space available

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/software/test/receive-test.py", line 8, in <module>
    can1.send(msg)
  File "/usr/local/lib/python3.5/dist-packages/can/interfaces/socketcan/socketcan_native.py", line 476, in send
    raise can.CanError("Transmit failed (%s)" % exc)
can.CanError: Transmit failed ([Errno 105] No buffer space available)
```

Here is my `/boot/config.txt` end of file : 
```
dtparam=spi=on
dtoverlay=mcp2515-can0,oscillator=16000000,interrupt=25
dtoverlay=mcp2515-can1,oscillator=16000000,interrupt=24
dtoverlay=spi0-hw-cs
```

And `/etc/network/interfaces.d/can` file : 
```
auto can0
iface can0 inet manual
  pre-up ip link set $IFACE type can bitrate 250000 listen-only off
  up /sbin/ifconfig $IFACE up
  down /sbin/ifconfig $IFACE down

auto can1
iface can1 inet manual
  pre-up ip link set $IFACE type can bitrate 250000 listen-only off
  up /sbin/ifconfig $IFACE up
  down /sbin/ifconfig $IFACE down
```

Thanks a lot for your help.

Olivier.

odenizot, 2018-05-17:
I send the frames to port 1 without having equipment connected to it.
I just noticed that unacknowledged frames remain in the transmit buffer, and therefore the buffer overflows.

Olivier.

felixdivo, 2018-05-22:
So this is a problem with SocketCAN or the CAN protocol itself, right?

odenizot, 2018-05-22:
Hello Felix,

I think that this problem only appears with PDU1 frames (peer-to-peer) because they need acknowledgement.
In my previous test, I used a timeout of 0.01 second, and I already had the problem.
When I read the documentation (2.1.0), I notice this information : "timeout (float) – If > 0, wait up to this many seconds for message to be ACK:ed or for transmit queue to be ready depending on driver implementation. If timeout is exceeded, an exception will be raised. **Might not be supported by all interfaces**.". I think with Vector or PiCAN2 it doesn't work.

wangdino, 2018-05-23:
Maybe play with the interface property `presume-ack on/off` ?

https://www.kernel.org/doc/Documentation/networking/can.txt

[image]

closed 2018-06-12 by felixdivo

Hector897, 2023-01-23:
I have the same problem of "No buffer space available". I was carrying out a severe investigation and I found the command "sudo ifconfig can0 txqueuelen 1000", when I execute it the error changes its name and now it appears as transmit buffer full" does anyone know how to solve it?

zariiii9003, 2023-01-23:
@Hector897 Please open a new issue and provide more information

Hector897, 2023-01-23:
I leave more information about it I need your help
https://github.com/hardbyte/python-can/issues/1495