hardbyte/python-can #314: Assertion error when trying to parse a legit CANoe generated BLF file

tburly, 2018-05-28:
I'm trying to parse a BLF using [this approach](https://stackoverflow.com/questions/49499903/how-do-i-convert-blf-data-from-can-to-csv-using-python):

```
import can
from pprint import pprint as pp

filename = "logfile.blf"
logging = can.BLFReader(filename)
logging = list(logging)
pp(logging)
```

and get 
`Traceback (most recent call last):
  File "D:\eclipse-workspace\blf2csv\pythoncan_blf2csv.py", line 7, in <module>
    logging = list(logging)
  File "C:\Users\my_user_name\AppData\Local\Python\Python36\lib\site-packages\can\io\blf.py", line 129, in __iter__
    assert header[0] == b"LOBJ", "Parse error"
AssertionError: Parse error`

The file is legit (I can convert it to CSV using Vector CANoe `Tool >> Log File Conversion` dialog (so I can convert a single file but looked at `python-can` as a means to do a batch conversion of multiple log files)).

felixdivo, 2018-05-28:
I do now know what that line should achieve, but as noted in #108 by @christiansandberg, finding documentation about that format is difficult ... Maybe he knows what's wrong with this.

tburly, 2018-05-28:
Additional info. 
When trying to parse a smaller file (on the same setup) I get a different error:
`Traceback (most recent call last):
  File "D:\eclipse-workspace\blf2csv\pythoncan_blf2csv.py", line 7, in <module>
    logging = list(logging)
  File "C:\Users\mazik\AppData\Local\Python\Python36\lib\site-packages\can\io\blf.py", line 122, in __iter__
    data = zlib.decompress(obj_data, 15, uncompressed_size)
MemoryError`
This file was recorded over a couple of seconds and has ca. 3.5MB. Trying the same code on a slightly weighter file (ca. 6MB) recorded on the same setup gives the above assertion error.

christiansandberg, 2018-05-28:
Try the master. I've done some changes since last release since the [reference implementation](https://bitbucket.org/tobylorenz/vector_blf) I've based my implementation on has made some changes regarding `uncompressed_size` that might cause memory issues.

christiansandberg, 2018-05-28:
Also try printing each message as they are read. Do you see any messages at all?

```python
import can

filename = "logfile.blf"
logging = can.BLFReader(filename)
for msg in logging:
    print(msg)
```

tburly, 2018-05-28:
Trying to read it one message at a time didn't change anything. But I'll try with the master branch and report back, thx.

felixdivo, 2018-05-28:
The `master` branch should be the same as the one on PyPi, please use the `develop` branch.

tburly, 2018-05-29:
Update: I tried `develop` branch - the same result.

christiansandberg, 2018-05-30:
It could be that the file header size has changed. Right now it's hardcoded to 144 bytes but it can be read from the file instead which would make it more robust to future changes. I'll try to fix that tonight.

christiansandberg, 2018-05-30:
Or actually I believe it is the usage of a newer object header (v2). I'll fix that as well.

closed 2018-06-05 by christiansandberg

jimmy1409, 2019-11-23:
Hey @christiansandberg  I am still getting this issue with Newest version 3.3.2
Should I checkout to some different commit to solve it?

christiansandberg, 2019-11-24:
@jimmy1409 would you be able to send a log file for me to analyze? You can e-mail me if you don’t want to upload it publicly.

B3NJ4M1N0, 2020-04-23:
Hi @christiansandberg I had also been getting this issue a few weeks ago and decided to use the MATLAB function blfread() to read blfs as I could filter the initial import of the file by the ID. My theory is that this made the file small enough to fit in memory (I'm working with .blf files that are up to 2gb in size and neither Python nor MATLAB could load the whole thing at once, even with 32gb of RAM). I was just wondering if there was something within the python-can library that would facilitate the same functionality or if not then I could suggest it for future development? Many thanks for the work that has been put in already.

christiansandberg, 2020-04-23:
File size should not be an issue since the file is parsed lazily (i.e. objects are only read when they are requested by the iterator) as long as you don't try to convert it to a list or something. But since it is Python it is not the fastest parser.

B3NJ4M1N0, 2020-04-27:
Sorry I tried to reply to this as soon as you commented but it wouldn't let me, and I haven't remembered until now! I had an enumerate function which must have been part of the issue, so I can start to read the BLF file successfully, however, I still eventually get the BLFParse error. I can send you a log via email if that would be helpful.

christiansandberg, 2020-04-27:
I did a fix last week in develop you can try out:

```
pip install https://github.com/hardbyte/python-can/archive/develop.zip
```

B3NJ4M1N0, 2020-04-27:
Just tried the fix, still seem to be getting a parse error around ~90,000 rows in however.

christiansandberg, 2020-04-29:
If you could send a file I could take a look: christiansandberg at me dot com.

christiansandberg, 2020-04-29:
I tried the file you sent with develop and it parsed without any errors for me. Maybe you need to check your installation. The can/io/blf.py file should contain this code on line 223:

```python
            # Find next object after padding (depends on object type)
            try:
                pos = data.index(b"LOBJ", pos, pos + 8)
            except ValueError:
                if pos + 8 > max_pos:
                    # Not enough data in container
                    return
                raise BLFParseError("Could not find next object")
```

B3NJ4M1N0, 2020-04-29:
Huh! Interesting, I'll give it a whirl tomorrow morning and make sure to
check that code. Thanks again for having a look.

Ben

B3NJ4M1N0, 2020-04-30:
Looks like it's working now, not sure why my first pip install didn't work..
Cheers

brotatos, 2022-06-16:
Has this fix been pushed to pypi? I am getting a similar error with my pip installation.

zariiii9003, 2022-06-16:
@brotatos Yes, this one was fixed. Please use the bug report template to create a new issue if you are using python-can 4.0.0.

PR #316 (2018-06-05): Add support for new BLF files
Supports version 2 of object headers.
Supports any file header size.
Better exceptions on parse errors.

Fixes #314.