hardbyte/python-can #320: Bad dlc on received messages

matisla, 2018-06-05:
I had error a message:
`WARNING:can.message:data link count was 12 but it should be less than or equal to 8`

I checked with CANalyzer what's happend:
[image]
But everything looks good.

after some test I see this:
`Timestamp: 1528206835.494456        ID: 03e9    S          DLC: 12    00 00 00 00 00 00 00 00`

This happend only on one message of my DBC.

How is it possible ? Where should I look to find the bug ?
I use a CANCaseXL from Vector

felixdivo, 2018-06-05:
Are you working with normal CAN or CAN-FD frames?

matisla, 2018-06-05:
Normal CAN 2.0A (11 bit arbitration ID)

felixdivo, 2018-09-06:
Well, apparently no one could help you as of now. Me neither, because I don't own a CANCaseXL. Have you been able to further locate the problem? Did you try using the C API directly?

felixdivo, 2018-10-09:
I will close this as we cannot reproduce this.

closed 2018-10-09 by felixdivo