hardbyte/python-can #331: PCAN Interface Error

CJKB, 2018-06-18:
I'm quite new to using python and am trying to read multiplexed CAN data from a separate unit. The way it works is by first sending a message to the unit to tell it to send back the configuration data in a different message. It works fine the first time I run the code (at In[1]), but then when I then try and run it again (In[2]), I get an error. There are no variables left over other than 'EEPROMData'. I was hoping that the 'bus.shutdown' command would fix this as I imagine that there's something open in the background which prevents the bus from initialising. Interestingly if I run this in debug mode rather than from the python console it works.

I'm using PyCharm and I'm running the script from the console. Here's my console input and the returned error. The code is below:

In[1]: execfile("Scratch_Script.py")
EEPROMData Read

In[2]: execfile("Scratch_Script.py")
Traceback (most recent call last):
  File "C:\anaconda\Anaconda3\lib\site-packages\IPython\core\interactiveshell.py", line 2963, in run_code
    exec(code_obj, self.user_global_ns, self.user_ns)
  File "<ipython-input-4-f7271f4922f5>", line 1, in <module>
    execfile("Scratch_Script.py")
  File "C:\Program Files\JetBrains\PyCharm Community Edition 2018.1.4\helpers\pydev\_pydev_imps\_pydev_execfile.py", line 18, in execfile
    exec(compile(contents+"\n", file, 'exec'), glob, loc)
  File "Scratch_Script.py", line 69, in <module>
    EEPROMData = canlistener()
  File "Scratch_Script.py", line 40, in canlistener
    bus = can.interface.Bus(bustype='pcan', channel='PCAN_USBBUS1', bitrate=1000000)
  File "C:\anaconda\Anaconda3\lib\site-packages\can\interface.py", line 87, in __new__
    return cls(channel, **kwargs)
  File "C:\anaconda\Anaconda3\lib\site-packages\can\interfaces\pcan\pcan.py", line 98, in __init__
    raise PcanError(self._get_formatted_error(result))
can.interfaces.pcan.pcan.PcanError: A PCAN Channel has not been initialized yet or the initialization process has failed

`   
   
    def canlistener():
        bus = can.interface.Bus(bustype='pcan', channel='PCAN_USBBUS1', bitrate=1000000)
        i = 0
        a_listener = can.BufferedReader()
        notifier = can.Notifier(bus, [a_listener])
        msg_array = []
        byteoffset = 0
        while i < 55:
            msg = can.Message(arbitration_id=0x214,
                          data=[22, 0, 0, 0, 0, 0, 0, 0],
                          extended_id=False)
            bus.send(msg)
            m = a_listener.get_message(None)
            if m is not None and m.arbitration_id == 0x213:
                msgdata = (list(m.data))
                if msgdata[0] == 15 and msgdata[1] == byteoffset:
                    i = i+1
                    byteoffset = i*4
                    print(byteoffset)
                    msg_array.append(msgdata)

        msg = can.Message(arbitration_id=0x214,
                      data=[0, 0, 0, 0, 0, 0, 0, 0],
                      extended_id=False)
        bus.send(msg)

        bus.shutdown
        print("EEPROMData Read")
        return msg_array

    EEPROMData = canlistener()

    
`

christiansandberg, 2018-06-18:
Try `bus.shutdown()` instead of `bus.shutdown`.

CJKB, 2018-06-19:
That has fixed it. Thanks very much!

closed 2018-06-19 by CJKB