hardbyte/python-can #349: Importing tries to load SocketCAN on Windows

tpwrules, 2018-07-04:
I just upgraded to 2.2.0. Now when I do `import can` on Windows, I get the following logging messages:
```
fcntl not available on this platform
libc is unavailable
```
This appears to be the result of trying to load SocketCAN, but that will never work on Windows and the error messages clutter up my program. These messages were not printed in 2.1.0.

christiansandberg, 2018-07-04:
Hmm right, it's because we've had `can.interface.CyclicSendTask` and `can.interface.MultiRateCyclicSendTask` that are SocketCAN only. Is anybody actually using these? I'm guessing it wouldn't be a big deal if we removed it. It's not really documented either. Just a commented out section in one of the examples with a note:

> Will have to consider how to expose items like this. The socketcan interfaces will continue to support it... but the top level api won't.

hardbyte, 2018-07-11:
Thanks for reporting Thomas. Yeah I think we can make a bugfix release for this.

PR #354 (2018-07-11): Only expose SocketCAN classes directly on Linux
Fixes #349.
closed 2018-07-11 by christiansandberg