hardbyte/python-can #352: Socketcan Listener Bufferedreader

pradeepdeeps, 2018-07-05:
Hi All,

I am looking for an example code that shows me how to add a listener and have my own function parse each message.

In short this is what I have been trying to achieve:
Every time there is a new message, the message should be queued to a buffer
The buffered messages should then be dequeued and processed.

If someone could point me to a sample code to achieve this would be a tremendous help.

Thanks
Pradeep

Bobet21, 2018-07-06:
Hello Pradeep, i understood the Python-can library should by default be buffered,however 
 I have been trying to get and process message iterating over the bus, but it appears that i am missing some (a significant number of) messages. So i would be interested in the same type of solution - probably using the class of "Listener".

christiansandberg, 2018-07-09:
The CAN hardware has some small buffer, then the driver or OS will have yet another buffer. The size of these buffers may differ so it is important to not stop reading new messages. This can be accomplished with the [BufferedReader](https://python-can.readthedocs.io/en/stable/listeners.html#can.BufferedReader) listener. Perhaps something like this:

```python
bus = can.Bus()
reader = can.BufferedReader()
notifier = can.Notifier(bus, [reader])

time.sleep(10)
# Process messages when suitable
msg = None
while msg is None:
    msg = reader.get_message()
```

Bobet21, 2018-07-10:
Hello Christian, 

Thanks for your response, that means that if the the time required to process the below code exceeds the buffer time we will miss messages? Is your time.sleep just there to show what would happen when processing messages? 
```
for msg in bus:
    try:
        df=pd.DataFrame(index=range(1),columns=range(10))
        df[0][0]=msg.timestamp
        df[1][0]=msg.arbitration_id
        candavec=list(msg.data)
        lengthdata=len(candavec)
        for i in range(0,lengthdata):
            df[i+2][0]=candavec[i]
        dfcan=dfcan.append(df)
    except KeyboardInterrupt:
#        dfcan.rename(columns={0:'Timestamp',  1:'Id',  2:'D1',  3:'D2',  4:'D3',  5:'D4',  6:'D5',  7:'D6',  8:'D7',  9:'D8'}, inplace=True)
        dfcan.set_index(pd.to_datetime(dfcan[0], unit='s'), inplace=True)
        dfcan.to_csv('dfcan.csv')
sys.exit()
```

christiansandberg, 2018-07-10:
`BufferedReader` is suitable when you want to do other blocking operations in your application and only process messages from time to time. That's what I was trying to exemplify with `time.sleep`. While your application is doing something else, the Notifier and BufferedReader will process the receive buffer for you.

In your case it might not make much of a difference since I don't see any blocking operations except maybe for some file writing? You still need to have enough CPU resources (on a single core since threads run on the same core) to both receive messages and process them.

closed 2018-07-18 by felixdivo