hardbyte/python-can #356: Unify exception handling

christiansandberg, 2018-07-10:
There is currently no portable way of knowing what has gone wrong in case of an error. Perhaps there should be some more specific subclasses of `can.CanError` for some generally usable exceptions.

If this sounds like a good idea we should start a list of exceptions that could be useful.

* Transmit buffer full
* Unknown channel
* Hardware not connected
* Connection/handle is down
* Bus off

felixdivo, 2018-07-18:
An Idea of categorizing this:

- CanError
  - CanInitializationError
    - unknown/invalid config/channel
    - drivers could not be loaded
    - wrong OS
    - device could not be found
  - CanOperationError
    - IO buffer full
      - send buffer full
      - receive buffer full
    - malformed message
    - cannot parse output of kernel/driver
    - some other error from kernel/driver

hardbyte, 2018-07-19:
Additionally we should specify how we capture any additional context that the various interfaces might make available. At the very least we usually have an interface specific error code which we should attach to the exceptions in a consistent way.

felixdivo, 2018-07-19:
Yes, as well as a human readable error message.

felixdivo, 2018-09-02:
Second Idea of categorizing this:

- `CanError`
  - generic top-level error for this library
  - `MalformedCanMessage`
      - invalid timestamp
      - illegal DLC field
  - `CanBusInitError`
    - unknown/invalid config/channel
    - drivers could not be loaded
    - current OS not supported
    - device could not be found
    - version/handshake problems
  - `CanOperationError`
    - cannot parse output of kernel/driver
    - some other error from kernel/driver
    - `CanBufferError`
      - send/receive buffer full

felixdivo, 2018-09-02:
What about errors that are specified by the CAN protocol itself? Are there some scenarios that are considered an error that we need to model with this?

hardbyte, 2019-02-23:
yeah I don't see why not

marcel-kanter, 2019-04-29:
How about:
- CanError
  - CanBackendError
     driver/OS/backend related stuff
    - drivers could not be loaded
    - current OS not supported
  - CanMessageError
     Message format/data specific errors
     - invalid timestamp
     - illegal DLC field
  - CanInitializationError
    Initialization specific errors:
    - unknown/invalid config/channel
    - device could not be found
    - version/handshake problems
  - CanOperationError:
    operation specific errors:
    - bit stuff error
    - form error
    - acknowledgment error
    - bit error
    - CRC error

marcel-kanter, 2019-04-30:
I've forked the repo and made a branch.
After having a look into message.py I think the errors for the message may all be handled with the built-in python exceptions, A beginning is in the _check method of the Message object already.

felixdivo, 2019-04-30:
@marcel-kanter Your structure looks nice, though I do not like the name `CanBackendError` since they are all kind of "backend errors". Maybe `CanDriverInitError`? And the current `CanInitializationError` renamed to `CanBusInitError`?

felixdivo, 2019-04-30:
Do we really want to include this still in 3.2? It could take some time to get it right. (@hardbyte)

marcel-kanter, 2019-04-30:
To the naming:
We (the contributors of python-can) sit in the middle and have two counterparts: First the user, to whom we expose the front-end - a set of python classes and methods with a certain level of abstraction. Secondly the driver and OS, which we use to provide the functionality and abstraction to the user.
Now call the part that is exposed to the user front-end and the other part the back-end.

The CanBackEndError is related to the backend of the python-can library, more or less the driver/OS level. It may be raised if the driver returns an unexpected value during operation, not only during initialization. 
One major reason for this may be a driver version (which the user installed) that does not exactly match the current implementation in python-can. This - i think - may be an even bigger issue on linux mashines where the drivers may be compiled by the user.
Additionally - I wanted to keep the words Driver, OS and Library out of the name.
I'm open for suggestions.

The CanInitializationError will become the CanBusInitError.

marcel-kanter, 2019-04-30:
> 
> 
> Do we really want to include this still in 3.2? It could take some time to get it right. (@hardbyte)

I think this may become an bigger change, since all interfaces need to be reviewed at least and some todos spread around...
Maybe for release 4.0?

felixdivo, 2019-05-01:
> I wanted to keep the words Driver, OS and Library out of the name.

OK, sounds good!

> The CanInitializationError will become the CanBusInitError.

This makes it clear that it is not a problem with loading the driver/whatever/... but instead of creating an instance of it. 👍

felixdivo, 2019-05-19:
> Should we add the return code as a parameter, which is None if not explicitly set? Also applies to CanInitializationError. This way, we could unify it a little bit, and it would be opt-in for interfaces where it would be helpful to have the error codes as a user.

felixdivo, 2019-05-19:
At first this will be developed on branch `unified-exceptions`.

marcel-kanter, 2019-05-24:
Since the exception handling comes with 4.0.0 release i removed the deprecations and the Python 2.7 support already. Who wants to integrate this already, have a look at Commit
@https://github.com/marcel-kanter/python-can/commit/3fea58867c1b1cbf075c7b0b7331a20031651154

felixdivo, 2019-06-23:
@marcel-kanter They are already removed on develop because of #594, #595, #596 and #597.

PR #1025 (2021-05-09): Add specific exceptions
This PR:
- adds specific base exceptions (they were discussed in #562 and before that in #356)
- documents these
- uses them in
  - all of the `can/*.py` modules
  - all of the `can/io/*.py` modules (it is not used there anyways, and that is correct)
  - the IXXAT interface (since this was already done in the original PR)
  - (This list excludes most of the interfaces. I propose doing them in separate PRs to keep the diffs and conflicts small. See [comment below](https://github.com/hardbyte/python-can/pull/1025#issuecomment-824973650).)

Closes #356. This PR is based on [... cut]
closed 2021-05-09 by mergify[bot]