hardbyte/python-can #358: Implementation of an Auto-Baud

Rittel, 2018-07-13:
Hey everyone!

I am currently considering to implement an Auto-Baud switching mechanism (125k, 250k, 500k, 1M Baud)
To do so I would listen on the Bus (maybe with peak-can in PASSIVE mode) and if there is traffic I would see if I am reading error-frames or "valid" messages. If I am only encountering error-frames I would adjust the Baud-Rate to another value chosen from a list of pre-defined rates. 

Has anyone already implemented anything similar or had any experience with this specific problem? If you have another suggestion on how to approach this problem I would be happy to hear that too.

Thanks already!

felixdivo, 2018-07-19:
Well, that basically sounds good. Maybe the PEAK driver can already do this and we would just have to call some API? If not, we might want to implement this in some more generic way, so multiple interfaces can benefit from it.

marcel-kanter, 2019-05-23:
The idea is good and should work. But Passive mode is essential elsewise you will disturb the messages on the bus.
But some of the professional use drivers (Ixxat, Vector) have this feature in their driver build in.
I would use the driver method first, if it does not exist, do it the manual way.