hardbyte/python-can #376: Specific exceptions for PCAN errors

altendky, 2018-07-30:
Somewhat related to #356 but more specific to PCAN.

Any interest in some error-specific exceptions?  I just wrote up this generator to create an enumeration (`enum.IntFlag`) of errors as well as individual exceptions for each (non-composite) error.  It also parses the comments from the source, but maybe that should be skipped in favor of calling the pcan interface for getting text for an error number.

I thought it would be nice to keep the source code modification to a minimum, but this could instead be a source generation script if we want real code in a file to provide these exceptions.

https://repl.it/@altendky/python-can-pcanbasic-exceptions-and-error-enumeration

<details> <summary>Click to expand</summary>

```python3
import enum
import inspect

import PCANBasic

def generate(module, prefix):
    all_errors = {
        name[len(prefix):]: getattr(module, name)
        for name in dir(module)
        if name.startswith(prefix)
    }

    compound_errors = {
        name: value
        for name, value in all_errors.items()
        if sum(int(c) for c in '{:b}'.format(value)) > 1
    }
    # print(compound_errors)

    simple_errors = {
        name: value
        for name, value in all_errors.items()
        if name not in compound_errors
    }

    Errors = enum.Flag('Errors', all_errors)

    source = inspect.getsourcelines(module)[0]
    comments = {
        getattr(Errors, line.partition('=')[0].strip()[len(prefix):]): line.partition('#')[2].strip()
        for line in source
        if line.startswith(prefix)
    }

    def create_exception(error):
        comment = comments[error]

        def __init__(self):
            print('yep, it is set.', comment)
            self.args = (comment,)

        return type(
            error.name + 'Error',
            (Exception,),
            {
                '__init__': __init__,
            },
        )

    exceptions = {
        getattr(Errors, name): create_exception(getattr(Errors, name))
        for name in simple_errors
    }

    return Errors, comments, exceptions

Errors, comments, exceptions = generate(
    module=PCANBasic,
    prefix='PCAN_ERROR_',
)
```
</details>

felixdivo, 2021-07-14:
Related to #1046.

felixdivo, 2022-02-06:
> Related to #1046.

I think this is sufficiently covered by that.

closed 2022-02-06 by felixdivo